MatterHackers/MatterControl #951: Slice settings 'Connection Info' missing from specific profiles

jlewin, 2016-06-15:
[image]

jlewin, 2016-06-27:
This issue was due to old profiles having a BaseLayer which was a snapshot of config.ini settings at the time of creation and those profiles lacked the field which caused the settings to appear in the SliceSettingsWidget. We recently modified the object model to no longer snapshot config.ini settings and instead we fall back to whatever defaults are in the build that you are running.

johnlewin, 2016-06-27:
- [ ] Confirm that these changes have the expected effect and that the bug is closed

gregory-diaz, 2016-06-30:
I just confirmed that this bug has been addressed but don't have write permissions on here. Feel free to move it to the completed list whenever you get a chance (John/Lars)

closed 2016-07-08 by larsbrubaker