MatterHackers/MatterControl #1035: Odd behavior relating to 'Require Leveling To Print'

rytz, 2016-07-01:
'Require Leveling To Print' hides the on/off switch in the Software Print Leveling section in OPTIONS, which is confusing and not the most efficient UI. Would like something more intuitive so users can make the connection between these two.

larsbrubaker, 2016-07-01:
The theory behind this is that if it is required it cannot be turned off. Do you have a suggestion or direction as to what might make it more intuitive. Some printers that require software leveling literally are not able to print without it so being able to turn it off only results is the printer not functioning.

rytz, 2016-07-01:
That makes sense, but I think the missing switch is what is confusing. In that case, I would do the following: 

- Having the switch still appear but be greyed out
- Incorporating help text (or a hover message) to indicate that another setting affects this switch

closed 2018-07-17 by jlewin