MatterHackers/MatterControl #1152: add new printer should pull the latest printer profiles.

larsbrubaker, 2016-07-21:


mmoening, 2016-07-21:
Adding a printer calls loadcacheable which will attempt to download the latest before defaulting to what is currently on cache or falling further back to static data

johnlewin, 2016-07-24:
MatterControl technically does this as it refreshes the "latest" printers list on startup and calls LoadCacheable at the end of the printer wizard and attempts to download the latest known profile for the selected printer - or falls back to cache, or falls back to StaticData.

To clarify, I believe this issue really means MatterControl should keep the "latest" printer profile definitions in sync, rather than letting them get stale after first load. This could potentially be hooked up in the sync process or we could perform this refresh when the wizard loads. Each has its trade offs.

closed 2016-08-07 by larsbrubaker