MatterHackers/MatterControl #1200: Presets "stash" changes on user layer that they are set to change

mmoening, 2016-07-27:
Currently if you have a user override set for a value and you apply a preset that changes that value the preset value is hidden under the user layer change.

It might be cool if when we apply a preset and we see that the user has an override for it that we stash it for when either the preset is set to default or the preset that was previously set

An alternative  is to make some sort a visual cue when a setting has a preset's value hidden below a user setting

larsbrubaker, 2016-07-27:
Seems like it might cause some confusion in the current model, but this is too big of a change for 1.6.

Post 1.6, I thin we should:
- [ ] If there is a preset setting set, that is currently a set user setting, pop up a wizard and ask the user if they want the preset. This will clear the user override for this setting.

larsbrubaker, 2017-12-03:
fixed

closed 2017-12-03 by larsbrubaker