MatterHackers/MatterControl #1249: Sli3er produces - nothing -

gamecompiler, 2016-08-05:
Hi, when Setting the Slice Engine to Sli3er, the produced gcode is "empty".

`; generated by Slic3r 1.2.9 on 2016-08-05 at 14:04:50

; external perimeters extrusion width = 0.25mm
; perimeters extrusion width = 0.56mm
; infill extrusion width = 0.56mm
; solid infill extrusion width = 0.56mm
; top infill extrusion width = 0.56mm

M107
M104 S210
M190 S60
M109 S210
G28
G1 Y90
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
; MatterControl Version 1.5.3 Build 1.5.3.6876 : GCode settings used
; Date 08/05/2016 00:00:00 Time 14:04
; avoid_crossing_perimeters = 1
; bed_shape = rectangular
; bed_size = 180,180
; bed_temperature = 60
; bottom_clip_amount = 0
; bottom_solid_layers = 3
; bridge_acceleration = 0
; bridge_fan_speed = 100
; bridge_flow_ratio = 1
; bridge_speed = 60
; brim_width = 0
; build_height = 180
; cancel_gcode = M104 S0\nM140 S0\n\nM4
; center_part_on_bed = 1
; complete_objects = 0
; connect_gcode = 
; cool_extruder_lift = 0
; print_leveling_solution = 3 Point Plane
; print_leveling_method = Manual
; print_leveling_required_to_print = 0
; manual_probe_paper_width = .1
; cooling = 1
; create_raft = 0
; default_acceleration = 0
; disable_fan_first_layers = 2
; end_gcode = M104 S0 ; turn off temperature\nM140 S0\nG91\nG1 E-5.0 F12000\nG90\nG28 X\nM84     ; disable motors
; external_perimeter_speed = 35
; external_perimeters_first = 0
; extra_perimeters = 1
; extruder_clearance_height = 20
; extruder_clearance_radius = 20
; extruder_count = 1
; extruder_offset = 0x0,0x0,0x0,0x0
; extruder_wipe_temperature = 0
; extruders_share_temperature = 0
; extrusion_axis = E
; extrusion_multiplier = 1.0
; extrusion_width = 0
; fan_always_on = 0
; fan_below_layer_time = 60
; filament_diameter = 1.75
; fill_angle = 0
; fill_density = 30%
; fill_pattern = honeycomb
; first_layer_acceleration = 0
; first_layer_bed_temperature = 75
; first_layer_extrusion_width = 0.4
; first_layer_height = 0.3
; first_layer_speed = 12
; first_layer_temperature = 205
; g0 = 0
; gap_fill_speed = 20
; gcode_arcs = 1
; gcode_comments = 0
; gcode_flavor = reprap
; gcode_output_type = REPRAP
; has_fan = 1
; has_hardware_leveling = 0
; has_heated_bed = 1
; has_power_control = 0
; has_sd_card_reader = 0
; show_reset_connection = 0
; heat_extruder_before_homing = 0
; include_firmware_updater = None
; infill_acceleration = 0
; infill_every_layers = 1
; infill_extruder = 1
; infill_extrusion_width = 0
; infill_first = 0
; infill_only_where_needed = 0
; infill_overlap_perimeter = 0.06
; infill_speed = 40
; infill_type = LINES
; layer_gcode = 
; layer_height = 0.25
; layer_to_pause = 
; max_fan_speed = 100
; min_extrusion_before_retract = 1.5
; min_fan_speed = 60
; min_print_speed = 5
; min_skirt_length = 2
; notes = 
; nozzle_diameter = 0.4
; only_retract_when_crossing_perimeters = 1
; ooze_prevention = 0
; output_filename_format = [input_filename_base].gcode
; overhangs = 1
; pause_gcode = G91\nG1 Z10 E-10 F12000\nG90
; perimeter_acceleration = 0
; perimeter_extruder = 1
; perimeter_extrusion_width = 0
; external_perimeter_extrusion_width = 100%
; perimeter_speed = 40
; perimeters = 2
; post_process = 
; print_center = 90,90
; raft_air_gap = 0
; raft_extruder = 0
; raft_extra_distance_around_part = 5
; raft_fan_speed_percent = 100
; raft_layers = 0
; raft_print_speed = 100%
; bed_remove_part_temperature = 0
; randomize_start = 0
; repair_outlines_extensive_stitching = 0
; repair_outlines_keep_open = 0
; resolution = 0
; resume_gcode = G91\nG1 Z-10 E10.8 F12000\nG90
; retract_before_travel = 5
; retract_layer_change = 1
; retract_length = 6.5
; retract_length_tool_change = 10
; retract_lift = 0.3
; retract_restart_extra = 0
; retract_restart_extra_toolchange = 0
; retract_when_changing_islands = 1
; retract_speed = 110
; skirt_distance = 3
; skirt_height = 1
; skirts = 2
; slowdown_below_layer_time = 40
; small_perimeter_speed = 30
; solid_fill_pattern = rectilinear
; solid_infill_below_area = 70
; solid_infill_every_layers = 0
; solid_infill_extrusion_width = 0
; solid_infill_speed = 60
; solid_shell = 0
; spiral_vase = 0
; standby_temperature_delta = -5
; start_gcode = M104 S210\nM190 S60\nM109 S210\nG28\nG1 Y90
; start_perimeters_at_concave_points = 0
; start_perimeters_at_non_overhang = 1
; support_air_gap = .3
; support_material = 0
; support_material_angle = 45
; support_material_percent = 4
; support_material_create_internal_support = 1
; support_material_create_perimeter = 1
; support_material_enforce_layers = 0
; support_material_extruder = 1
; support_material_extrusion_width = 0
; support_material_infill_angle = 30
; support_material_interface_extruder = 1
; support_material_interface_layers = .9mm
; support_material_interface_spacing = 0
; support_material_pattern = honeycomb
; support_material_spacing = 4
; support_material_speed = 40
; support_material_threshold = 0
; support_material_xy_distance = 0.7
; support_material_z_distance = 0.15
; support_material_z_gap_layers = .3mm
; support_type = LINES
; temperature = 210
; thin_walls = 1
; threads = 4
; toolchange_gcode = 
; top_infill_extrusion_width = 0
; top_solid_infill_speed = 50
; top_solid_layers = 3
; travel_speed = 300
; use_firmware_retraction = 0
; use_relative_e_distances = 0
; vibration_limit = 0
; wipe = 0
; wipe_shield_distance = 0
; wipe_tower_size = 0
; z_can_be_negative = 0
; z_offset = 0
`

When using the CuraEngine as SliceEngine, the preview is empty (But gcode is still correct <- works).

[image]

Thanks in advance,
gamecompiler

unlimitedbacon, 2016-08-05:
Could you post your [slice settings file](http://wiki.mattercontrol.com/Send_Debugging_Information#Slice_Settings) and the STL file? Are you running 1.5.3 stable for compiling from source? 

See also [this information on debugging Slic3r](http://wiki.mattercontrol.com/Frequently_Asked_Questions#.22Slicing_Error.22_When_using_Slic3r).

gamecompiler, 2016-08-05:
Im using software version 1.5.3
My MatterControl settings with Sli3er activated are:
Id not compile from source, id downloaded the current release from MatterControl @ http://www.mattercontrol.com/

```
avoid_crossing_perimeters = 1
bed_shape = rectangular
bed_size = 180,180
bed_temperature = 60
bottom_clip_amount = 0
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 60
brim_width = 0
build_height = 180
cancel_gcode = M104 S0\nM140 S0\n\nM4
center_part_on_bed = 1
complete_objects = 0
connect_gcode = 
cool_extruder_lift = 0
print_leveling_solution = 3 Point Plane
print_leveling_method = Manual
print_leveling_required_to_print = 0
manual_probe_paper_width = .1
cooling = 1
create_raft = 0
default_acceleration = 0
disable_fan_first_layers = 2
end_gcode = M104 S0 ; turn off temperature\nM140 S0\nG91\nG1 E-5.0 F12000\nG90\nG28 X\nM84     ; disable motors
external_perimeter_speed = 35
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_count = 1
extruder_offset = 0x0,0x0,0x0,0x0
extruder_wipe_temperature = 0
extruders_share_temperature = 0
extrusion_axis = E
extrusion_multiplier = 1.0
extrusion_width = 0
fan_always_on = 0
fan_below_layer_time = 60
filament_diameter = 1.75
fill_angle = 0
fill_density = 30%
fill_pattern = honeycomb
first_layer_acceleration = 0
first_layer_bed_temperature = 75
first_layer_extrusion_width = 0.4
first_layer_height = 0.3
first_layer_speed = 12
first_layer_temperature = 205
g0 = 0
gap_fill_speed = 20
gcode_arcs = 1
gcode_comments = 0
gcode_flavor = reprap
gcode_output_type = REPRAP
has_fan = 1
has_hardware_leveling = 0
has_heated_bed = 1
has_power_control = 0
has_sd_card_reader = 1
show_reset_connection = 0
heat_extruder_before_homing = 0
include_firmware_updater = None
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 0
infill_overlap_perimeter = 0.06
infill_speed = 40
infill_type = LINES
layer_gcode = 
layer_height = 0.25
layer_to_pause = 
max_fan_speed = 100
min_extrusion_before_retract = 1.5
min_fan_speed = 60
min_print_speed = 5
min_skirt_length = 2
notes = 
nozzle_diameter = 0.4
only_retract_when_crossing_perimeters = 1
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
pause_gcode = G91\nG1 Z10 E-10 F12000\nG90
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0
external_perimeter_extrusion_width = 100%
perimeter_speed = 40
perimeters = 2
post_process = 
print_center = 90,90
raft_air_gap = 0
raft_extruder = 0
raft_extra_distance_around_part = 5
raft_fan_speed_percent = 100
raft_layers = 0
raft_print_speed = 100%
bed_remove_part_temperature = 0
randomize_start = 0
repair_outlines_extensive_stitching = 0
repair_outlines_keep_open = 0
resolution = 0
resume_gcode = G91\nG1 Z-10 E10.8 F12000\nG90
retract_before_travel = 5
retract_layer_change = 1
retract_length = 6.5
retract_length_tool_change = 10
retract_lift = 0.3
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_when_changing_islands = 1
retract_speed = 110
skirt_distance = 3
skirt_height = 1
skirts = 2
slowdown_below_layer_time = 40
small_perimeter_speed = 30
solid_fill_pattern = rectilinear
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extrusion_width = 0
solid_infill_speed = 60
solid_shell = 0
spiral_vase = 0
standby_temperature_delta = -5
start_gcode = M104 S195\nM190 S60\nM109 S195\nG28\nG1 Y90
start_perimeters_at_concave_points = 0
start_perimeters_at_non_overhang = 1
support_air_gap = .3
support_material = 0
support_material_angle = 45
support_material_percent = 4
support_material_create_internal_support = 1
support_material_create_perimeter = 1
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_infill_angle = 30
support_material_interface_extruder = 1
support_material_interface_layers = .9mm
support_material_interface_spacing = 0
support_material_pattern = honeycomb
support_material_spacing = 4
support_material_speed = 40
support_material_threshold = 0
support_material_xy_distance = 0.7
support_material_z_distance = 0.15
support_material_z_gap_layers = .3mm
support_type = LINES
temperature = 195
thin_walls = 1
threads = 4
toolchange_gcode = 
top_infill_extrusion_width = 0
top_solid_infill_speed = 50
top_solid_layers = 3
travel_speed = 300
use_firmware_retraction = 0
use_relative_e_distances = 0
vibration_limit = 0
wipe = 0
wipe_shield_distance = 0
wipe_tower_size = 0
z_can_be_negative = 0
z_offset = 0
```

unlimitedbacon, 2016-08-05:
The problem is that you have the Use Arcs option turned on in Settings > Printer > Features > Firmware (`gcode_arcs = 1`). [Slic3r does not support this feature](https://github.com/alexrj/Slic3r/issues/23) unless you specifically enable it when compiling. We should remove this option from the UI.

@larsbrubaker, when you run Slic3r from the command line, it gives the following message:

    Arc fitting is not available (incomplete feature)

Can we PLEASE start displaying these kind of messages? It would make issues like this far easier to diagnose, both for us and for the users.

larsbrubaker, 2016-08-05:
Removed in next build

closed 2016-08-05 by larsbrubaker