MatterHackers/MatterControl #1253: import .ini file seems to be non-functional

larsbrubaker, 2016-08-05:


larsbrubaker, 2016-08-08:
May need more info from Ryan. Initial investigation did not show the problem.

jlewin, 2016-08-09:
@rytz - Do we have any details regarding this problem? Was there an ini file that could be attached to this ticket that might expose the problem behavior and/or was this issue simply a manifestation of a bad build?

rytz, 2016-08-09:
After review, I think this problem may be different than I originally described. The .ini file I used was probably incorrect as it was my attempt to recreate settings from a previous print by using a .gcode file and saving only the settings at the end as its own .ini file. So the variables are wrong for MatterControl which is probably why it won't load. 

However, the fact that it won't load or isn't compatible isn't clear by looking at the UI after the attempt to open the file is made. This is confusing, and I think an error message would make the failure much more apparent.

Link: https://drive.google.com/open?id=0B0ofelzlgYFPNVQySjA5eVZJNjg

johnlewin, 2016-08-09:
After reading through your experience and running through the import process with the linked file it sounds like the following work should be completed to improve the user experience:

_"Warn users if imported files contains no known settings"_

I've changed the title of the bug to reflect the desired behavior. Please advise if that doesn't seem right but doing so would inform the user that the import action they just completed had no effect and might have identified the problem with less effort

**Original Title**
import .ini file seems to be non-functional

johnlewin, 2016-08-09:
@rytz  - fyi, if you want to skip a step in the future, you can natively attach files to GitHub issues by dragging them into the editor

rytz, 2016-08-09:
@johnlewin That all looks great-- thanks. And I tried attaching... GitHub doesn't like .ini files.

jlewin, 2016-08-09:
Gotcha. I think Lars had some files he renamed or zipped the other day for this exact reason. Thanks for the reminder

closed 2016-08-22 by larsbrubaker