MatterHackers/MatterControl #1282: "Support Everywhere" not functioning

cope413, 2016-08-12:
Support everywhere doesn't change anything on this part.
[image]

larsbrubaker, 2016-08-12:
Can you attach the stl and settings?

unlimitedbacon, 2016-10-22:
A similar issue seems to be occurring here.

[image]

Support columns are added even though they cannot extend down to the bed.

Slice Settings, STL, and GCode:
[SupportEverywhere.zip](https://github.com/MatterHackers/MatterControl/files/545487/SupportEverywhere.zip)

larsbrubaker, 2017-12-03:
fixed

closed 2017-12-03 by larsbrubaker