MatterHackers/MatterControl #1295: Presets pencil icon remains clickable when no preset is selected

rytz, 2016-08-16:
For Material and Quality presets, the pencil icon next to the drop down menus is meant to be a quick-access button for editing the currently-selected preset. If no preset is selected, however, the pencil icon does not change-- it remains clickable and illuminates when moused over. 

I suggest that the icon is greyed out or otherwise shown to be unusable unless a preset is selected. Furthermore, it seems that changing '- default -' to '- none -' or similar might help the user more quickly make the connection between a greyed-out button and not being able to edit.

larsbrubaker, 2016-08-17:
The pencil is currently not opening the window but still enabled. It would be better if the pencil was disabled when menu to default.
- [ ] Disable pencil button when set to default item.

larsbrubaker, 2017-12-03:
Working

[image]

closed 2017-12-03 by larsbrubaker