MatterHackers/MatterControl #1303: Export to Zip puts all files in a folder called "project-assembly"

unlimitedbacon, 2016-08-18:
If you Export to Zip from the Queue menu, the zip file will contain a folder called "project-assembly" which has all the files in it. It seems like this folder should have a more logical name or not exist at all.

jlewin, 2016-08-19:
Seems too late to change as existing clients are hard-coded to that path

jlewin, 2016-08-19:
After some reflection, it seems more context is needed. A number of features in the product rely on a common implementation that creates or extracts MatterControl generated zip files. The "Send" feature in particular would break in older clients if we changed the structure of the zip file and that's what my original comment was speaking to, where for some parts of the product we're locked into the current format for compatibility reasons.. or so it seems at first glance

johnlewin, 2018-09-24:
Nothing had changed since this big was originally filled. We continue to push zip content into the weirdly named sub directory