MatterHackers/MatterControl #1326: Clicking 'Done' after print sometimes doesn't remove Queue item which just printed

rytz, 2016-08-26:
Not exactly sure how to repro this bug, but I know it happens. 

*Might have to do with opening a model in new window using 'View' button while other model is printing.  Not sure.

Will update this report as I find more information.

jlewin, 2016-08-26:
**Repro**

1. Open MatterControl
2. Switch to the Queue tab
3. Start a short print
4. Click the View button on any non-active queue item to popup a new window
5. Wait for the print to complete
6. Once completed, click the Done button

**Expected**
The printed item would be removed from the queue

**Actual**
The item which was viewed in the popup window is removed

larsbrubaker, 2017-12-03:
No longer part of our workflow (2.0). Closing.

closed 2017-12-03 by larsbrubaker