MatterHackers/MatterControl #1403: Add the possibility to load a file into MatterControl from commandline

nokianhunter, 2016-09-19:
I don't know if the present program structure will allow you to implement this but it would be really comfortable.
Imagine typing on the commandline or invoking this command from other programs: 
"mattercontrol MyCube.stl"
This will add the .stl file to the MatterControl Queue and you can go on slicing it directly.

larsbrubaker, 2016-09-19:
Have you tired this? It should already be working.

nokianhunter, 2016-09-19:
Yes I have tried on Linux. I discovered that you can type whatever you want after "mattercontrol" and it won't get recognized. A command overview when typing "mattercontrol --help" would be helpful even if it just says: "Usage: mattercontrol [FILE]"

unlimitedbacon, 2016-09-19:
Confirmed. Although it works when building from source, it doesn't work when installing the Debian package. This is because `/usr/bin/MatterControl` is actually a Bash script that calls MatterControl.exe indirectly and doesn't pass on the command line arguments.