MatterHackers/MatterControl #1406: Strange raft/skirt issue in Experimental

cope413, 2016-09-19:
Irregular skirt and raft shapes being generated in EXP build. 

Objects are 2 perfect spheres.

[image]

larsbrubaker, 2016-09-21:
It seems as though the problem with this is, the mesh has a highly dense area on its side. As this is only a problem with the support outline, we will leave this for the future.

[image]

jstefanop, 2016-09-22:
Are there daily builds that can be downloaded somewhere? Im still on the build from May and hoping for something newer soon.