MatterHackers/MatterControl #1536: Settings to new window.. Auto closing..

jatmn, 2016-10-25:
When Settings is opened to a new window it is auto closing on its own randomly.. right now it appears to be random some times it does it with a moment after clicking a text box.. other times its doing it after clicking a few text box's.. Its generally closing after imputing a new value. 
*I have not been hitting tab or return.. it just closes after I am done typing or after I click to a new text box.

Version 1.6.0
Windows 8.1 64bit

tellingmachine, 2016-10-26:
> thread[2].

I noticed that too.

Links:

  1. https://github.com/MatterHackers/MatterControl/issues/1536
  2. https://github.com/

larsbrubaker, 2018-09-24:
Closed - fixed.
Please re-open if this is still a problem.

closed 2018-09-24 by larsbrubaker