MatterHackers/MatterControl #1545: Crash while launching second instance while first instance is printing

unlimitedbacon, 2016-10-27:
This happens about 50% of the time.

```
Unhandled Exception:
SQLiteUnix.SQLiteException: database is locked
  at SQLiteUnix.SQLite3.Prepare2 (System.IntPtr db, System.String query) [0x00027] in <51e6fdbffa7d4ae586059b9e3a8e7b28>:0 
  at SQLiteUnix.SQLiteCommand.Prepare () [0x00011] in <51e6fdbffa7d4ae586059b9e3a8e7b28>:0 
  at SQLiteUnix.SQLiteCommand.ExecuteNonQuery () [0x00012] in <51e6fdbffa7d4ae586059b9e3a8e7b28>:0 
  at SQLiteUnix.SQLiteConnection.Execute (System.String query, System.Object[] args) [0x00040] in <51e6fdbffa7d4ae586059b9e3a8e7b28>:0 
  at SQLiteUnix.SQLiteConnection.CreateTable (System.Type ty) [0x000b0] in <51e6fdbffa7d4ae586059b9e3a8e7b28>:0 
  at MatterHackers.MatterControl.DataStorage.Datastore.ValidateSchema () [0x00019] in <51e6fdbffa7d4ae586059b9e3a8e7b28>:0 
  at MatterHackers.MatterControl.DataStorage.Datastore.Initialize () [0x0001c] in <51e6fdbffa7d4ae586059b9e3a8e7b28>:0 
  at MatterHackers.MatterControl.MatterControlApplication.Main () [0x00054] in <51e6fdbffa7d4ae586059b9e3a8e7b28>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: SQLiteUnix.SQLiteException: database is locked
  at SQLiteUnix.SQLite3.Prepare2 (System.IntPtr db, System.String query) [0x00027] in <51e6fdbffa7d4ae586059b9e3a8e7b28>:0 
  at SQLiteUnix.SQLiteCommand.Prepare () [0x00011] in <51e6fdbffa7d4ae586059b9e3a8e7b28>:0 
  at SQLiteUnix.SQLiteCommand.ExecuteNonQuery () [0x00012] in <51e6fdbffa7d4ae586059b9e3a8e7b28>:0 
  at SQLiteUnix.SQLiteConnection.Execute (System.String query, System.Object[] args) [0x00040] in <51e6fdbffa7d4ae586059b9e3a8e7b28>:0 
  at SQLiteUnix.SQLiteConnection.CreateTable (System.Type ty) [0x000b0] in <51e6fdbffa7d4ae586059b9e3a8e7b28>:0 
  at MatterHackers.MatterControl.DataStorage.Datastore.ValidateSchema () [0x00019] in <51e6fdbffa7d4ae586059b9e3a8e7b28>:0 
  at MatterHackers.MatterControl.DataStorage.Datastore.Initialize () [0x0001c] in <51e6fdbffa7d4ae586059b9e3a8e7b28>:0 
  at MatterHackers.MatterControl.MatterControlApplication.Main () [0x00054] in <51e6fdbffa7d4ae586059b9e3a8e7b28>:0 
```

larsbrubaker, 2016-10-27:
This is due to the sqlite database being locked for the history % update.

larsbrubaker, 2018-12-29:
The new build will only run one instance and connect to multiple printers. Making fixed.

closed 2018-12-29 by larsbrubaker