MatterHackers/MatterControl #1553: Can't select printer immediately after importing

unlimitedbacon, 2016-10-28:
Steps to reproduce:

1. Import a .printer file as a new printer
2. Select the new printer from the list
3. The selection will go to "Printers..." instead of to the printer you wanted
4. It works when you try the second time

larsbrubaker, 2016-10-31:
If it changes from the current printer to 'printers...' that would be a bug, but not selecting the imported printer is the intended behavior.

larsbrubaker, 2016-10-31:
Seems to be doing the expected behavior in my testing. I'll come look at it on your setup.

unlimitedbacon, 2016-10-31:
Happens with any of these profiles.
[Profile Import.zip](https://github.com/MatterHackers/MatterControl/files/562869/Profile.Import.zip)

larsbrubaker, 2016-11-03:
I cannot reproduce this on windows.

unlimitedbacon, 2016-11-03:
Do you think it has anything to do with the fact that I always have a yellow connection indicator? You can use my system for debugging if necessary.

jlewin, 2016-11-07:
It looks like this is functioning with the attached profiles on the originally offending machine. Not sure if it's a fluke or not. Tyler will take a look soon to confirm.

closed 2016-11-07 by unlimitedbacon