MatterHackers/MatterControl #1598: "Purchased" folder does not appear in Library

unlimitedbacon, 2016-11-08:
Seems to be intermittent. Restarting MatterControl sometimes fixes the problem.

larsbrubaker, 2016-11-08:
Could this be part of the proxy problem? Sometimes it takes a moment for to come up, but I am always seeing it.

closed 2016-11-08 by larsbrubaker

unlimitedbacon, 2016-11-08:
The proxy is disabled. John has seen this happen as well. He asked me to report it.

jlewin, 2016-11-08:
The condition appeared after the proxy settings were removed and the application was restarted but even if the was still in place, if anything had network access, everything had network access. The network was functioning as the yellow status icon was displayed.

jlewin, 2016-11-08:
This doesn't need to be a blocking issue, just a record of observed behavior