MatterHackers/MatterControl #1602: Z Offset should reset to 0 after running Software Print Leveling calibration

rytz, 2016-11-08:
Z Offset, if different than 0, is guaranteed to be incorrect after running the wizard for 3-, 7-, or 13-point calibration for Software Print Leveling. 

Z Offset should automatically be set to 0 after this wizard completes so any previous entries do not affect the next print.

larsbrubaker, 2016-11-08:
This is a very complicated feature to implement. And needs much consideration.

This could be done by having a common set of settings that are cleared at the beginning of print leveling. We should consider what kind of messaging will be delivered to the user. We should also think if there are any other potential confounds.

larsbrubaker, 2017-01-24:
We also need to reset the new baby stepping z offset.

jlewin, 2017-01-24:
>This could be done by having a common set of settings that are cleared at the [**beginning**] of print leveling

What if before I start leveling things are working well but I'd like to improve them. I start leveling but midway through decide it's too much work and cancel. Ideally since I canceled I'd be back at the state where I started, not reset to some new unprintable and different values. 

I assume we're clearing at the beginning of the wizard to get to raw machine values during the wizard moves but it seems like that could be accomplished in non-invasive ways and allow us to only reset values at the successful completion of the wizard.

rytz, 2017-01-24:
> that could be accomplished in non-invasive ways and allow us to only reset values at the successful completion of the wizard.

The negative effect of clearing at the beginning has come back to bite me before when I accidentally cleared my settings by hitting the CONFIGURE button.

closed 2017-02-06 by larsbrubaker