MatterHackers/MatterControl #1631: Want more presets for filament density

rytz, 2016-11-17:
Under the SETTINGS -> Material -> Density, the only presets are for PLA and ABS. We have data on many more, and they should be integrated into the program. 

http://matterhackers.helpscoutdocs.com/article/16-how-long-is-a-spool-of-filament

unlimitedbacon, 2016-11-17:
I have a comprehensive list of filament densities here:
https://docs.google.com/a/matterhackers.com/spreadsheets/d/1ImJCOo0dX6H4kGsp6FwB07wp87DmixQyr8BDDMtFLTM/edit?usp=sharing

rytz, 2016-11-17:
@unlimitedbacon Even better.

larsbrubaker, 2016-11-17:
I do not want to populate this list with filaments that the user cannot print. I think we should put densities, into the material list for a specific printer. We can probably add one or two materials here (the default list) if you want, but more than that we should put into the materials list.

unlimitedbacon, 2016-12-21:
I'm going to add PET to the list. @rytz, do you think any other materials would be appropriate?

rytz, 2016-12-21:
I think PET is the only other one that I would add given the comment @larsbrubaker made about not adding materials that most printers cannot print. I think PET is close enough to a standard material that its inclusion is justified.