MatterHackers/MatterControl #1641: If Share Temperature is enabled, the second extruder (if active) controls do not show in CONTROLS -> Movement

rytz, 2016-11-23:
If Extruder Count is set to 2 and Share Temperature is on, only one set of controls shows under the Movement section in CONTROLS. Both sets of extruder controls should show.

larsbrubaker, 2016-11-28:
Share Temperature means that both extruders are heated by one heating element. They are on the same block and and can't be set individually.

rytz, 2016-11-28:
> Share Temperature means that both extruders are heated by one heating element.

Right, but the temperature functionality isn't what's wrong. Separate temperatures shouldn't be able to be set individually, but they should be able to be extruded separately. The controls for individual extrusion are missing.

closed 2016-12-12 by larsbrubaker