MatterHackers/MatterControl #1643: No Proxy Configuration options

frankvdh, 2016-11-23:
So I can't login or get updates or any other network thing from inside MatterControl.

larsbrubaker, 2016-11-28:
Not sure what you mean. What network information are you trying to get? Or are you just having trouble logging into your account?

gmoyle, 2017-02-03:
I think the question is regarding users who try to run the software in a corporate environment with a proxy. There are no options in the software to allow configuring the software to work with a proxy, so the user is unable to login. I am experiencing the same issue. The app is not proxy friendly.

jlewin, 2017-02-04:
MatterControl relies on proxy information defined using Microsoft's common .NET application configuration file (MatterControl.exe.config) in our install directory. 

On my machine this file is located at:
>C:\Program Files (x86)\MatterControl\MatterControl.exe.config

You can edit file on disk and match your environment needs. By default we ship configured to use the system defined proxy but I don't believe that supports autopac setups which may account for the behavior you're seeing. Our default proxy config is:
```xml
<system.net>
  <defaultProxy>
    <proxy bypassonlocal="false" usesystemdefault="true" />
  </defaultProxy>
</system.net>
```

You can try setting the `proxyaddress' attribute of the `proxy` element to the ip and port of your proxy and MatterControl should then be configured to know about your proxy:
```xml
<system.net>
  <defaultProxy>
    <proxy bypassonlocal="false" proxyaddress="http://[your proxy address and port number]" />
  </defaultProxy>
</system.net>
```