MatterHackers/MatterControl #1644: Lost all my filament presets after update

Penged, 2016-11-24:
They were all there for the first print. I didnt close matter control but when I started my next print all my filament presets were gone. It has all the generic settings. Im really unhappy about this.

unlimitedbacon, 2016-11-25:
Sorry that the migration to the new version did not work well for you. Fortunately your old settings are not lost, you can still get them back.

[Here are instructions](http://matterhackers.helpscoutdocs.com/article/76-manually-migrating-materials-from-15-to-16?auth=true).

Penged, 2016-11-27:
When I click on the link it asks me for an email address and password for helpscout. Is there a different link that will take me to the instructions?

unlimitedbacon, 2016-11-28:
Sorry about that. Try again.

Penged, 2016-11-29:
Thank you. I was able to access it that time and it looks like I got everything back. It might be a good idea to let others learn from my mistake prior to updating mattercontol.