MatterHackers/MatterControl #1645: Calibration missing, Autocalibration not working

tjaekel, 2016-11-25:
New version 1.6 - the calibration wizard is not there anymore (and all my settings lost).
How to calibrate my Rostock V2?
The macro Autocalibration fails completely and is very risky !
It moves so much into bed, tries to dive into glass, sounds and looks very bad. It keeps moving and will move outside the bed after while.

Sorry, I am lost:
- new version has deleted all my setting !
- new version does not let me calibrate the printer !
- some settings, e.g. material as Custom, are lost and changed back when other menus were visited
Looks to me:
new version 1.6 made me unable to use my printer, to print anything. "Great", not well done!

unlimitedbacon, 2016-11-25:
Sorry. It appears that the wrong printer profile was being used for the Rostock v2 (was the same as the one for the v3). The v3 has autocalibration and the v2 does not, so obviously the autocalibration macro would not work for your printer.

This has been fixed. Delete your printer and add a new one, and it should pull down the correct profile this time.

tjaekel, 2016-11-27:
Thank you: YES, installing the printer again - it works.
No, calibration is back and the auto-calibration macro is gone.
Almost fine: just strange, why using the 13 point calibration has "small" issues: on some locations, e.g. 121.24 | 70, one arm goes so deep that is sounds strange and I cannot trim. The arms go on some locations deeper as the towers allow.