MatterHackers/MatterControl #1646: Settings not taken - cannot print anymore

tjaekel, 2016-11-25:
"Thank you" for the new version 1,6: I CANNOT print anything anymore!
If I set Extruder Temperature, e.g. 240 degC, when I start print temperature jumps back to 225 and jobs waits to cool down.
There is no save menu, save context anymore!
How to store my settings?

The new version 1.6 does not seem to be usable, sorry.
It will destroy existing settings, calibrations and does
not let me print anything anymore.

gamecompiler, 2016-12-14:
Got the same problem.
When i set the bed temp in the settings menu, the menu settings it self are saved!
But when i export a new gcode, the bedtemp is not set like my settings....
Id not encounter this problems on the old versions....
This problem occurs in Matterslice, Cura and Sli3er at the same.

My main gcode heat setup is as following:

``

M104 S[first_layer_temperature]
M190 S[first_layer_bed_temperature]
M109 S[first_layer_temperature]

``

If i unterstood gcode correctly (if not please say it) the slicer should insert my settings there. But it does not! I still got my old settings inserted, even if i restart Mattercontrol:

``

;Generated with Cura_SteamEngine 14.03
; automatic settings before start_gcode
G21 ; set units to millimeters
M107 ; fan off
T0 ; set the active extruder to 0
; settings from start_gcode
M104 S235
M190 S95
M109 S235

``

unlimitedbacon, 2016-12-14:
Looks like you are using the wrong variables in the Start G-Code. Should be...
```
M104 S[temperature]
M190 S[bed_temperature]
M109 S[temperature]
```
Here is a [list of supported variables](http://wiki.mattercontrol.com/Custom_G-Code_Variables).