MatterHackers/MatterControl #1647: V1.6 does not let me print anymore

tjaekel, 2016-11-25:
The new version 1.6 does not let me save settings. Even settings changes, e.g. extruder temperature - they are not taken.
After installing this version - I cannot print anymore. I am unable to calibrate printer, unable to start printing with my settings etc. Version is not usable with Rostock Max V2.

larsbrubaker, 2016-11-28:
We have corrected the V2 printer profile to allow calibration. Just create a new printer and it should have calibration enabled.

Settings are saved every time they are changed in the new version. There is no longer a save button. If you are logged into your account you should be able to see your most recent changes by going to settings history. I've included a screen shot to help show where this is.

[image]

tjaekel, 2016-12-08:
OK, thank you.
But to be honest: every new version is worse than before. The new update V1.6.1.7967 drives me crazy (new ticket filed - you are killing your business  👎 ).