MatterHackers/MatterControl #1648: V1.6 MatterSlice infill overlap wrong

tjaekel, 2016-11-25:
Using the default slicer "MatterSlice" - I see now, it prints diagonal (not rectangular). OK, fine,
would work better for my models.
But how to control the printing orientation?

The Infill Overlap has an issue:
on the left hand side it looks fine, reasonable overlap between perimeter and infill. But on the right hand
side it is wrong: there is a large gap between inside infill and perimeter.
And it is not parallel: on the top right part much larger gap, at the lower right the gap gets smaller, but
still there. On left hand side it looks similar: more overlap on the top left part.

It looks like: the infill pattern is slightly rotated, not really parallel with the rectangular perimeter.

unlimitedbacon, 2016-11-25:
We have not seen this issue on any of the models we have tested. Could you upload the STL, G-Code, and your slice settings?
http://wiki.mattercontrol.com/Send_Debugging_Information#Other_Useful_Files

You can always change the angle of the infill so that it is not running parallel to the perimeters. The setting is in Settings > General > Infill > [Starting Angle](http://wiki.mattercontrol.com/Settings/General/Infill/Advanced/Starting_Angle). We normally use 45 degrees.

tjaekel, 2016-11-27:
Thank you.
I will try to upload all requested files. The issue with the gap between perimeter and infill is still there, even printing something else. On one side (e.g. bottom) it is OK, on upper side a large gap.