MatterHackers/MatterControl #1649: Just upgraded to 1.6.0 -- Tried a print and although it puts the correc temp in when you use a profile, when you export it and load it to the printer, the temperature sent is incorect.

JMMcMahan, 2016-11-27:


larsbrubaker, 2016-11-28:
Can you attache the exported profile that your used to generate the gcode? Will investigate without, but it might help.

Thank you.

JMMcMahan, 2016-11-29:
Mr. Brubaker –

 

Thanks for the response. 

 

Attached is the .stl that I put into MatterControl to do the slice.

Attached is the PET Profile that I used – it is the same that I have been using with the old MC version (1.5.0) I have been using successfully.

I created the drawing and saved the .STL from TurboCad Professional 2016.

I used Matter Slice to do the slicing.

 

I have deleted the new MC Version and the gcode files that failed and gone back to the old so do not have a copy of the resultant gcode that did not set the temps correctly, sorry. 

 

Hope this helps.

 

Thanks,

John

larsbrubaker, 2016-12-03:
Hey John, I'm sorry to say but I did not get the attachment sent through email. Can you add them as a zip through github. Thanks in advance.

tjaekel, 2016-12-08:
YES, for me as well: with latest updates - my extruder does not heat anymore!
And I see often value 0 as target (on LCD or Windows tool), even jumping temperature values between 0 and target (or set value and target) on Windows tool.
And if I see my print result: it looks like the temperature (shown) is completely wrong (PET printed with 250 degree C but it looks like a much lower temperature used).

gamecompiler, 2016-12-14:
Got the same problem. 
Got Mattercontrol 1.6 and Windows7 SP1

[DaVinci.zip](https://github.com/MatterHackers/MatterControl/files/652637/DaVinci.zip)

unlimitedbacon, 2017-02-01:
@gamecompiler, the problem is with your [Start G-Code](http://wiki.mattercontrol.com/SETTINGS/Printer/Custom_G-Code/Start_G-Code). The variables you are using to set the temperatures are `[first_layer_temperature]` and `[first_layer_bed_temperature]`. Instead they need to be `[temperature]` and `[bed_temperature]`. See the [list of custom G-Code variables](http://wiki.mattercontrol.com/Custom_G-Code_Variables).

Everyone else, I can't tell without seeing your printer profiles but I suspect it is also an issue with your Start G-Code. This issue will be closed unless you want to provide more information.

JMMcMahan, 2017-02-01:
Tyler –

 

Thank you for the response. I set the temperatures in the same fields I use in V1.5 which are labelled “Extruder Temperature” and “Bed Temperature” and it works correctly with V1.5 (sets the nozzle and bed temperatures correctly when I save the gcode on the SD card and select that print from the printer controller. I did not specifically set the ‘first layer temperature’ for either nozzle or bed so cannot say exactly what it was using. I will retry the testing I did and wait for it to finish the first layer and then see if it changes to the Extruder Temperature and Bed Temperature that I had entered in the profile.

 

Thanks again for looking into this – I will let you know what I find in the test.

 

-John

unlimitedbacon, 2017-02-22:
@JMMcMahan, I don't think you read my previous messages very carefully. I think the issue is with your Start G-Code. This wiki page will show you where to find it.
http://wiki.mattercontrol.com/SETTINGS/Printer/Custom_G-Code/Start_G-Code

The Start G-Code uses variables in brackets to insert the appropriate settings in the correct place. I believe that, like @gamecompiler, your G-Code either has the wrong variables or is missing them entirely. You can read more about the custom G-Code variables here.
http://wiki.mattercontrol.com/Custom_G-Code_Variables

It would be very helpful if you could send us your slice settings file so we can examine your Start G-Code ourselves. However, you will need to do this by uploading it through Github, not as an email attachment. Here are instructions on exporting your slice settings.
http://wiki.mattercontrol.com/Send_Debugging_Information#Slice_Settings