MatterHackers/MatterControl #1657: Ultimaker 2 Profile crossing perimeter or not retracting

alecgates15, 2016-11-29:
Looking at the gcode, it does look like the reason I was getting some bad prints on the Ultimaker is, for whatever reason, it is crossing perimeters or not doing retractions when it should.
[Bugged U2 Profile.zip](https://github.com/MatterHackers/MatterControl/files/620105/Bugged.U2.Profile.zip)
[image]
[image]

alecgates15, 2016-12-01:
Also, default extrusion multiplier is 1.38. Not sure if that is intentional or not

unlimitedbacon, 2016-12-01:
Fixed extrusion multiplier in the profile.

alecgates15, 2016-12-09:
Profile:
[Ultimaker 2 Profile.zip](https://github.com/MatterHackers/MatterControl/files/643131/Ultimaker.2.Profile.zip)
File:
[Bracer_Right_Main_MIddle.zip](https://github.com/MatterHackers/MatterControl/files/643132/Bracer_Right_Main_MIddle.zip)

larsbrubaker, 2017-01-18:
Thank you.

This is fixed in master and the new [experimental](https://mattercontrol.appspot.com/downloads/matterhackers-experimental-windows/development) build.

closed 2017-01-23 by larsbrubaker