MatterHackers/MatterControl #1667: Odd Behavior Using "Touching Loop" vs. "First Layer Expansion" w/ Raft

Humanzee, 2016-12-01:
This appears to be several interactions between creating/not creating a raft, setting a number of priming loops and whether they're touching or not, and setting a number of expansion loops, so I'm not precisely sure of what happens with all of the combinations. this is on a Rostock Max v3 and MC 1.6.0.7828

But normally, setting touching priming loops behaves identically to setting expansion loops and creates a brim. However, if there is a raft, "priming loops" appear concentric, non-toughing loops and "expansion loops" do not generate anything at all.

larsbrubaker, 2017-01-18:
Internal note: My be fixed in master. After verify mark bug or fixed.

FireflyNJ, 2017-01-23:
I had some trouble with brim too.
In other software I am familiar with a setting to increase extruder flow on first layer to help adhesion.
In MatterControl (1.6.2.8214) we have a setting 'First Layer Extrusion Width'
I am not sure of the intention of this feature, it does effect the 'width' but why?, it does not help with adhesion as flow would. It has a negative effect on Brim (first layer expansion) it increased the width that it does not overlap and separates instead of holding the object. In fact I got better result with a lower number. see:
[image]
100% 120% and 200%

I couldn't see the changes looking at preview, it didn't refresh after hitting 'Generate'
I have to try the touching priming loop again, I didn't use it since it didn't touch object enough to hold it.

unlimitedbacon, 2017-01-24:
@Humanzee,

This is a case we had not considered. What do you think should happen when using first layer expansion (brims) with a raft?

Priming loops (skirt) and first layer expansion (brim) are actually quite different. Priming loops are intended to get the plastic flowing through the nozzle before it goes to print anything important. Priming loops encircle every object on the bed as if you had stretched a rubber band around all of them.

First layer expansion is for helping hold things to the bed. The loops are created around each object individually. I'm not sure what good it would do to use them in conjunction with a raft. Is it your intention to put them on the raft to help it stick better to the bed? Or do you want to put them around the base of the object to help it stick better to the raft?

@FireflyNJ,

The purpose of first layer extrusion width is to help with bed adhesion. The idea being that fatter lines will stick to the bed better. In our experience this has always worked well. Increasing the flow without increasing the line spacing does not make sense. If you did this, you would be overextruding and the lines would overlap each other.

Also, please do not hijack other people's threads. Your question is unrelated to Humanzee's original discussion about the interaction between rafts, brims, and skirts.

FireflyNJ, 2017-04-03:
I agree the purpose of the first layer extrusion width is to help with bed adhesion, but it breaks the brim, see the gaps in photo above, it just falls apart, no longer holding the part.

larsbrubaker, 2017-04-06:
Thanks for the pictures. 

It is clear that there should be a different behavior as we increase the first layer extrusion width. Making as bug and putting in icebox (for future development).

larsbrubaker, 2018-12-29:
This should all be fixed now.

closed 2018-12-29 by larsbrubaker