MatterHackers/MatterControl #1673: V1.6.0.7828 - slicer "slic3r" prints outside bed

tjaekel, 2016-12-02:
OK, I saw a new update: obvious, the latest one before seems to have quite a lot of issues ( which I saw).
I've installed.
I want to print a vase and have issues always at the same layer (approx. layer 23mm / 0.2 mm = approx. layer 115). It stops extruding, looks like clogging, I can hear on filament feed (it jumps backwards).
OK, I tried different slice engines:
a) MatterSlicer: it is very slow ! (soooooo slow !): it takes a lot of minutes to process my STL file. OK, but it shows the same trouble for my model (it messes up the print up after 1 hour on a certain layer)
b) I tried to use **"slic3r"**: much faster, print starts much faster (it seems to process the STL file much faster).
But: **it prints outside the bed !** It does not print in the bed center, instead: it starts printing in the right upper corner and moves outside the bed. Very bad!
c) I use the slicer "CuraEngine" right now, lowered the feed - let's see what happens.

My conclusion: pushing a new version of MatterControl after two weeks (I was quite disappointed about 
the previously released version - even it is obvious and reasonable to provide a new version), now a new version but still trouble ... it lets me assume: not all features with different models and setting combinations tested properly ...

Please, make sure, if you announce a new version then it is working better as a previous one.
Please, test all feature which users could configure, including setting combinations, e.g. using a different
slice engines. Thank you.
(my feeling is: I am going "backwards" with new versions: I had a MatterControl version, assuming 1.5
before, which was working quite well, but the new versions create trouble and hinder me printing).

Sorry, for harsh words, I want to print, and not evaluate versions which do not work well at the end.
Best regards
Torsten

unlimitedbacon, 2016-12-02:
This id a duplicate of bug https://github.com/MatterHackers/MatterControl/issues/773.

Fishbowlkraken, 2016-12-07:
It is, though it seems no resolution's been made. I really do want to upgrade to the new MatterControl, but I'm stuck on version 1.4 until they resolve this.

tjaekel, 2016-12-08:
Meanwhile - I agree with you: fall back to older version, don't go with latest version and updates (I have filed new trouble ticket: latest V1.6 update is even worse - all is going "backwards").

mmoening, 2016-12-08:
Hey I wanted to let you guys know that there is a work around for this if you would like. 
This bug is caused by us upgrading the Slic3r that we ship with MatterControl. This allows the use of a large number of additional features that wouldn't otherwise be supported. Unfortunately this does have the side effect of having a problem with one setting in particular as Slic3r has changed what values Slic3r expected without maintaining backwards compatibility. 
That being said if you replace the current slic3r folder with the one from the previous version's the bed center should be in the proper location, this does mean you lose support of the "new" features that are in the newer slic3r though.

We are currently working on a generic solution that will allow us to quickly and easily address this and similar issues that may arise in the future.

Fishbowlkraken, 2016-12-10:
It seems like the easiest solution would be to just extend the configuration file to allow users to set the bed dimensions using the legacy option. As it stands, being stuck with a version of Slic3r from _2014_ is kind of a big problem. I also noticed that in the Linux version of MC 1.6, Slic3r support was removed entirely, which, again, necessitates my staying on 1.4 for the time being. Honestly, I would appreciate some way to more easily add other slicers to MC's interface in general, as I'd love to try out Prusa's custom build of Slic3r with new infill and support options, as well as possibly trying out the more experimental Bread slicer, allowing for non-planar layers. As it stands, though, without a decent bit of scripting and modifications, this really is not possible. 

I really do love MatterControl, it's by far my favorite control suite for delta printers, but these issues really are making it difficult for me to use as my primary application.

unlimitedbacon, 2016-12-12:
The lack of Slic3r in the 1.6 package was an oversight. You can still add it yourself, though. Here are instructions.
http://wiki.mattercontrol.com/Development/Running_on_Linux#Using_Other_Slice_Engines

Fishbowlkraken, 2016-12-14:
You said yourself it was an oversight, why not release a new version of MatterControl with Slic3r integrated again? Having to manually set symlinks to restore core functionality really is not acceptable. Is making a point release for Linux really a lot to ask?

mc71, 2017-01-05:
I've found an apparent workaround on 1.6.
If you slice a part with Slic3r and the toolpaths show it off the bed, move the part and save it, then problem goes away.... or at least for me.