MatterHackers/MatterControl #1683: Dual Extrusion Bug in 1.6

SGTGamerAMP, 2016-12-03:
When printing from the second extruder ONLY for a single piece print worked fine to begin, then I canceled the print due to an issue but when i regenerated it it showed it printing from the first extruder, trying to print and it indeed tries to print form the first extruder. Unable to revert it back, tried restarting the program and computer. Worked around it by adding a calibration cube and resizing the z to .2 mm then it worked fine.

larsbrubaker, 2017-01-18:
Thanks, We'll look into this and create next steps.

unlimitedbacon, 2017-02-01:
Can you confirm that this is still happening on version 1.6.2? If so, go to Help > About MatterControl and click Clear Cache at the bottom. Then go back to the Layer View and click Generate. Does this have any effect?

larsbrubaker, 2017-02-21:
Thanks for the feedback. Closing - assuming this is fixed. Please let us know if not.

closed 2017-02-21 by larsbrubaker