MatterHackers/MatterControl #1707: Can't change defaults

whosawhatsis, 2016-12-08:
In version 1.6.1, I noticed that I have a setting that is wrong in my "default" profile for a machine (minimum travel requiring retraction was accidentally set to 20 instead of 2, due to general interface bugginess of the mono framework and me not paying enough attention), and now when I try to fix it, mattercontrol only seems willing to treat this as a temporary change, and there doesn't seem to be a way to make it the new default. Is there an override somewhere that I'm missing, or has the new version been locked down so thoroughly that I'll have to modify the file manually to fix it?

nokiann, 2016-12-08:
@whosawhatsis: At the moment there only is a workaround for this, by editing the profiles manually
Have a look here:
http://forums.matterhackers.com/topic/385/how-to-change-defaults-after-1-6-help

I have one question (after reading your post here http://forums.reprap.org/read.php?156,311834,page=2):
How do you get accurate holes nowadays? Do you use your own code?

whosawhatsis, 2016-12-08:
I never bothered with a library, I just type the r / cos(180/n) math wherever I need it. I do use low-poly holes whenever it really matters (6 or 8 facets). This produces overhangs no greater than 45 degrees (30, in the case of 6 facets), and is a simpler and more consistent version of the truncated teardrop hole shape from the early days.

If I really want the hole to be circular, I set $fs = .2; and $fa = 2; (actually, I usually do this globally anyway, and just don't override it for the hole). This makes the facets about half a nozzle width, making the holes very round without adding wasted segments that will only slow the machine down. These circles didn't come out well in the days of that post, but the technology has come a long way. If I want something to fit in one of these circular holes, I either add some tolerance to the diameter beforehand or ream the hole after printing (a drill bit running in reverse works great).

nokiann, 2016-12-08:
The low-poly holes are for horizontal holes. What about vertical holes?
What is "$fs = .2; and $fa = 2"?
So you have a manual approach and this feature is still missing in most slicing software.

whosawhatsis, 2016-12-08:
Those are openscad settings for how curves are converted to facets. I do this for vertical and horizontal holes, because it makes it so much easier to make things fit together. A nice octagonal vertical hole requires very little thought, and will fit a circle matching its apothem every time.

larsbrubaker, 2016-12-09:
You can edit the file manually, but if you have a change to a profile that should be part of the general system we can upload it to our database and it will update on any new printer creation. The tools that do the migration from user settings to default settings are currently only in our upload scripts.

closed 2016-12-12 by larsbrubaker