MatterHackers/MatterControl #1709: Version 1.6.1 upgrade loses 13-point calibration

teejaydub, 2016-12-08:
I upgraded to MatterControl 1.6.1.7967 via the 'stable' update channel, using it with an Orion v2. It lost my 13-point calibration, wanted me to do it again, I did it, then printed something. Went to print again, it wanted me to calibrate again. I deleted the printer and recreated it, after which it thought an Orion printer had hardware calibration - which no version of the Orion has never had. That also blew away all my saved settings - curated over months to years for many materials and quality options - which was a surprise. If you can tell me where those files were stored, I'd like to get them back from backups and move back to a previous version.

teejaydub, 2016-12-08:
Downloading and reinstalling 1.5.3 reconnected me to my existing settings and calibration, so my immediate problem is solved.

unlimitedbacon, 2016-12-08:
Ok I fixed the Orion profile and shut off hardware leveling. Here is a guide on migrating your settings from 1.5 to 1.6. Sorry but this will not get the calibration data. You will still have to redo that.
http://matterhackers.helpscoutdocs.com/article/76-manually-migrating-settings-from-15-to-16

teejaydub, 2016-12-08:
Thanks very much, I'll work on that. Or maybe think about scripting it... I have 24 Quality settings and 23 Material settings. Where is the calibration data stored in both versions?

unlimitedbacon, 2016-12-08:
In version 1.6, all of your settings (including leveling data) are stored in .printer files (which are really .json files). These are kept in `~/.local/share/MatterControl/Profiles/<accountname>`.

In the previous versions (1.5 and earlier), it was all stored in the SQLite database `~/.local/share/MatterControl/MatterControl.db`.

teejaydub, 2016-12-09:
Awesome, thanks! I don't know how soon I'll get to it, but if I script the conversion I'll make it available.

closed 2016-12-12 by larsbrubaker