MatterHackers/MatterControl #1711: MatterControl V1.6 update - every new version is worse

tjaekel, 2016-12-08:
I am sorry to say this: I have updated to new version, build 1.6.1.7967 - and the **list of trouble gets longer**. It looks like: every new version is **worse than before**. Sorry.

Short list of trouble:
1. used slicer "CuraEngine" (used because "MatterSlicer" has trouble with Infill Overlap, is too slow to
    parse the STL file ...) - nozzle moves down, short before touching is moves to Home and says:
    "Print Done" -
    pretty cool   👎 **I cannot print anything anymore with this slicer** (which was OK before).
2. The **extruder does not heat** very often: target displayed but sensor says it cools down
3. Cancel print: it does not come back to the buttons (it hangs)
4. It can even happen that the **entire application hangs** and you have to use the
    Task Manager in order to kill it.
5. Some **buttons disappear**, e.g. the "Set" button for the extruder temperature. The bad thing:
    even restarting all these buttons remain disappeared.
6. If I use "Home Printer" - it does, but after a while it goes back to bed, crashes into bed and tries to
    continue with the print.
    **"Home" during a running print lets crash the nozzle into bed!**

My suggestion (again):
before you release a new version - test it properly. Make sure you are not going backwards or make clients disappointed (like me). Make sure, people can still print (I cannot anymore, all broken).
You start jeopardizing your reputation and business.

Here the long list of issues for the SW developers of MatterControl:
a) **"CuraEngine" does not print**: moves down, back to home and "Print Done"
b) **"MatterSlice"** is quite **slow** to parse STL file and ready to print; it has **issues on Infill Overlap**
     (not symetrical, perimeter not connected to infill on one side)
c) The movement from Home to bed when starting the print is not smooth: it sounds and looks very
    strange (stopping in between, "stuttering")
d) To Home nozzle before starting a print is now earliefr - not known when
    (long time before the temperatures have reached the target - why changed?)
e) often, esp. after "Cancel" and "Print" again - the **extruder is not heated**
f) The **estimated printing time is completely wrong**: often it is done much faster,
   e.g. 1/2 of the estimated duration needed
g) Often I have situation that Target Temperature toggles between 0 degree and target.
    The **LCD can show also Target Temperature as 0**. It is very confusing on the extruder temperature.
h) I have used Target as 265 degree and "Set" for 260 degree: at the end, the Windows App and the LCD
    toggles between 260 and 265 and it does not print.
    And the "Set" button disappears and remains disappeared when starting all over.
i) I click on "Cancel" for the print - it does not come back: the buttons remain disappear, it keeps going
   to calculate the estimated time - no way to control anything.
h) I had several cases where **entire MatterControl hangs**, nothing can be done, no buttons accepted,
    Windows "Not Responding".
    It has to be killed with Task Manager. (never seen before !)
k) Also **"Disconnect"** (in order to do a new "Connect") can result in **hanging entire tool**.
l) If I change flow on local LCD when print was launched but not yet started on printer, e.g. to 80%,
   the speed jumps also to 80%: **changing the flow ratio changes also the speed ratio**.
m) If I **Home the printe**r during a running print - it moves back to bed after a while. But it is very bad:
    it **crashes into bed!** Doing it several times - the arm movement speed is different (gets slower).
n) On my print, after 2 layers done - the nozzle moved down again and crashed into bed !
o) I have now also trouble to heat the extruder: if I set 260 or 265 degree - it almost never reaches this
    temperature (I was able before to print PC with 290 degree). Now - I wait endless.
    I block the fan on extruder just to heat up properly (stop with finger - risky !).
    It is now almost **impossible to print above 250 degree extruder temperature** - it never reaches the
    temperature. Why he extruder fan is nott controlled?
p) New version has asked me to save my setting to cloud - I did (hoping with calibration as well).
    But printing now with previous setting (calibration, 13-point settings) - the bed does not look leveled:
    It is not parallel anymore - one side is way too low. Why? - nothing changed on mechanical trim.
    **Saved calibration does not seem to match anymore** w/o changes.
q) If I realize with "Print" (again) the extruder is not heated (even target is shown) - if I press "Preheat"
    even this does **not heat the extruder**. All has to be closed, power-cycle the printer and start over
    (also for every new print of the same file when done !)

So, my conclusion: MatterControl V1.6 and esp. the updates are not working, make it almost impossible
to print. Please, let me go back to V 1.4 or 1.5 (which was working fine before quite well, but never
versions are getting worse). The latest version is so flawed, almost impossible to be used. Sorry.

(just) regards
Torsten

tjaekel, 2016-12-08:
forgotten:
r) below the Windows app says: "checking for version" (of the printer FW): it does not display any
   reasonable FW version info there.

tjaekel, 2016-12-08:
forgotten:
s) older version of MatterControl, e.g. 1.5.3 connect automatically on application start.
   Newer version (1.6.1) does not - I had to print "Connect", older does automatically to find my printer.
   (clear indication of "going backwards" with your versions 👎

finchm, 2016-12-12:
I downloaded a slightly newer version than the reported  1.6.1.7967.  I had massive issues as well.

The skirt was drawn (and printed) through the center of one of my parts.  

Another of my parts was printed as a deformed mess (twice).  I backed off to 1.5.3 and all is well.

duisenberg, 2016-12-13:
1.6.x is still Alpha, isnt it? So it is not recommended for production use. These Alpha/Beta Versions are published as SAR-Versions - Search for errors and report them.

finchm, 2016-12-13:
If 1.6 is Alpha, then there is a huge bug in version 1.5.  

At the top of the screen on version 1.5 it says "Update Available".

Clicking on the text "Update Available" brings up a dialog box that says "There is a recommended update available".

My "Update Channel" is "Stable"

Clicking on the "Download Update" button downloads the garbage that is masquerading around as version 1.6

gamecompiler, 2016-12-14:
Had the same. Got now the "Stable" version 😠 ....