MatterHackers/MatterControl #1725: Printer Control: Z homing with AutoLevel enabled

duisenberg, 2016-12-13:
In Controls window with printer settings Autolevel checked, the Homing - Z button either should issue a G29 command or get a set of XY coords to go to before sending G28 Z.  

A Custom G-Code for Axis homing would do as well.

unlimitedbacon, 2016-12-13:
Normally, moving to a safe position before homing the Z axis is handled by the printer's firmware. Here is the config option in Marlin:
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Configuration.h#L872

duisenberg, 2016-12-13:
I run an a bit older Marlin 1.0 which is recommended by printer manufacturer, but already got the autolevel feature. Dont know if the current 1.0.2 has it changed.

The behavior I have with safe Z homing is if I use G28, it homes X and Y, then sets the safe position and homes Z. If I use G28 X Y and then G28 Z the first command is executed normally but the second results into a positional warning.  

I meanwhile wired a tiny relais parallel to the mechanical Z endstop that is triggered by the probe and disabled Z_SAFE_HOMING. This works quite well from mechanical point of view since the endstop catches the Z axis even if the probe is outside the bed.

duisenberg, 2016-12-15:
The old 1.0 used to print in spirals with autolevel, 1.02 printed but without any visible leveling effect, so I updated to 1.1.0 RC8 today. Other than the old versions it now drives the probe to center position when
probing Z individually. Though, after a few seconds (stepper auto release) it doesnt do anything until XY are homed again. Which is quite as it is meant to work.

I guess a note at autolevel checkbox help that individual homing might be affected by the autolevel management of the firmware would be a good thing. Everything else - well, since the autolevel feature
in firmware atm is subject of change it might be better to wait a bit until things have settled.

closed 2016-12-15 by duisenberg