MatterHackers/MatterControl #1754: OSX 10.12.2 crashing on launch

PyRo1509, 2016-12-22:
Stupid me I just upgraded OSX and now it MatterControl Doesn't run.

unlimitedbacon, 2016-12-22:
Could you follow the instructions here to gather a crash log?
http://wiki.mattercontrol.com/Send_Debugging_Information#Crash_Logs

PyRo1509, 2016-12-22:
```
/Applications/MatterControl.app/Contents/MacOS/MatterControlMac ; exit;
objc[22452]: Class VCWeakObjectHolder is implemented in both /System/Library/PrivateFrameworks/AVConference.framework/Frameworks/ViceroyTrace.framework/Versions/A/ViceroyTrace (0xa4bc8090) and /System/Library/PrivateFrameworks/AVConference.framework/Versions/A/AVConference (0xa4bdee98). One of the two will be used. Which one is undefined.
Fontconfig error: Cannot load default config file
Request Failed: No Active Firmware
Error CodeNotModified : Message - Not Modified
```

It took a while to come up with those messages.  Now it loads.

unlimitedbacon, 2017-02-01:
We have seen that on occasion MatterControl will take an excessively long time to start on a Mac. Usually this is due to an issue with Mono's font cache. The Fontconfig error in your log indicates that this is the case. See issue https://github.com/MatterHackers/MatterControl/issues/788.

Making sure you are on the right version of Mono may help. MatterControl currently works best with Mono 4.4.2. If you have any other versions on your system, it may help to uninstall them.

larsbrubaker, 2017-02-02:
Closing as dupe (it is referenced in the other bug now)

closed 2017-02-02 by larsbrubaker