MatterHackers/MatterControl #1755: 1.6.2: Serial command seq fail due to line no error

duisenberg, 2016-12-22:
Nice to see a xmas update:)

But ... MC 1.6.2 (8094) comes with a serious serial communication issue:  Command line number sequence error.

This is how it looks like in Terminal: 

[image]

The quick fix is to Reset Connection (which needs to be enabled in printer features).

didli, 2017-01-05:
Exact same problem here with same workaround, but with MatterControl 1.6.1 in Linux. The nozzle didn't crash into the bed but it's way off (starting gcodes not working), and there's no extruder or bed temp readings until I reset connection. Reverting back to 1.6.0 seems to works just fine.

unlimitedbacon, 2017-01-10:
Another user running into the same problem. See last post.
http://forums.matterhackers.com/topic/414/the-printer-stops-printing-when-printing-bridges/11

duisenberg, 2017-01-11:
It might be an interim fix to add two checkboxes to 1.6.2: one for using line numbers, one for checksum. 
This would help to trace the issue and continue to use the alpha versions. Even if there is a good chance for the issue being related to Marlin. Else would be sad to be cut of from using the new MC versions until by chance this is solved one day.

duisenberg, 2017-01-11:
To catch up with things ... the issue already is present using 1.6.1 (Windows) but for some reason MC and Marlin manage to negotiate that and stay in sync. See attached serial protocol dump.  
[print_log.txt](https://github.com/MatterHackers/MatterControl/files/698549/print_log.txt)

So disabling line numbers and/or checksums might not be the best idea since even if it would bypass the issue it has some serious odds. The initial handshake as done in MC1.6.1 seems to work well, so guess the best option is to revert to "the old" procedure.

duisenberg, 2017-01-11:
Might it be that Marlins M110 is classically simply ignoring its parameters and starts by default with line one expecting N2 to be the next line with a command?

`->Communication State: PreparingToPrint
->N1 M110 S1*96
<-ok
->M105
<-ok T:208.1 /195.0 B:45.9 /45.0 @:0 B@:0
->M105
<-ok T:207.3 /195.0 B:45.8 /45.0 @:0 B@:0
->Communication State: Printing
->N2 G21*24 [0.018]
`
(printer log as taken from 1.6.1 starting a print.  )

If so, it should be quite easy to fix the 1.6.2 issue simply by starting with N2 as first line after M110 instead of N1.

duisenberg, 2017-01-13:
The issue is related to [Marlin](https://github.com/MarlinFirmware/Marlin/issues/5677), present I guess since May 2016 where M110 has been subject of change. 

In order to use Marlin 1.1.0 RC8 with Mattercontrol 1.6.2 Marlin_main.cpp line 5658 
```/**
 * M110: Set Current Line Number
 */
inline void gcode_M110() {
  if (code_seen('N'))  gcode_N = code_value_long(); 
}
```
has to be changed to 

```if (code_seen('N')) { gcode_N = code_value_long(); gcode_LastN = gcode_N; }```

Tough - there still is an issue with Mattercontrol 1.6.2 using line numbers:
```
<-ok T:12.2 /0.0 B:13.0 /0.0 @:0 B@:0
->N49 M105*26
<-ok T:12.2 /0.0 B:13.0 /0.0 @:0 B@:0
->N50 M105*18
<-ok T:12.2 /0.0 B:12.9 /0.0 @:0 B@:0
->N51 M105*19
<-ok T:12.2 /0.0 B:12.8 /0.0 @:0 B@:0
->N52 G28*36 
<-echo:busy: processing
<-echo:busy: processing
<-echo:busy: processing
<-echo:busy: processing
->N52 G28*36
<-Error:Line Number is not Last Line Number+1, Last Line: 52
<-Resend: 53
<-ok
<-echo:busy: processing
->N53 M105*17
<-echo:busy: processing
<-echo:busy: processing
<-echo:busy: processing
<-echo:busy: processing
<-echo:busy: processing
->N53 M105*17
<-Error:Line Number is not Last Line Number+1, Last Line: 53
<-Resend: 54
<-ok
->N54 M105*22
<-echo:busy: processing
<-X:129.00 Y:94.00 Z:10.78 E:0.00 Count X: 10320 Y:7520 Z:4312
<-ok
<-ok T:11.7 /0.0 B:12.8 /0.0 @:0 B@:0
<-ok T:11.7 /0.0 B:12.8 /0.0 @:0 B@:0
->N55 M105*23
<-ok T:12.4 /0.0 B:13.3 /0.0 @:0 B@:0
->N56 M114*20
```
Seems to be the timeout for commands like G28 is a bit low.

unlimitedbacon, 2017-01-14:
The latest alpha version (1.6.2.8214) [should be fixed](https://github.com/MatterHackers/MatterControl/commit/642b6ad907ed1404e2e708da0c28fb3010e764aa) to handle this. Can you try it?

duisenberg, 2017-01-14:
Printer log snippet is from 1.6.2.8214 which I switched to directly after uploading the patched RC8 to the machine.  So it already contains the referenced commit.   

The lasting issue seems to come from re-injecting a command while Marlin still is processing it, As shown this happens with G28 (and also with G29 at print start), too in some rare cases within a print - like 5 times for 100k+ commands. 

Might be in PrinterConnectionAndCommunication.cs adding a callback for "echo:busy" to increase the timeout waiting for ok and to delay M105 could help to avoid unnecessary resends. 

Not sure how #1787 comes in here, but a look into the printer log will help to see what is happening there.

unlimitedbacon, 2017-01-18:
Just to clarify, 1.6.2.8214 should have fixed your original issue with the M110 command. Can you verify that it is working now with an unpatched version of Marlin? 

I am much less concerned about the issue with G28 taking too long to process. Although the situation is not ideal, there should actually be no negative effect since the printer ignores the extra commands with the same line number. If it is actually causing problems during printing, can you open another issue? I think it is best to have a single thread for each bug.

duisenberg, 2017-01-18:
Reuploaded an unpatched Marlin to the machine ... yes, I can positively confirm that 1.6.2.8214 also prints with the unpatched Marlin 1.1.0 RC8. 

The premature resend actually does not seem to cause major issues. Agree its best to open a separate issue for that, so this is #1832 by now.

closed 2017-01-18 by duisenberg