MatterHackers/MatterControl #1757: Dual extrusion with Shared Temp unable to select material

tkoopman, 2016-12-27:
Unable to pick which material to use when configuration has 2 extruders but single temp. If I un-tick single temp then option to pick material reappears.

Running latest 1.6.1 version

unlimitedbacon, 2016-12-27:
Isn't this the intended behavior? If you have a shared heater, you cannot print both materials at different temperatures. Therefore you must use the same or very similar materials in both nozzles.

tkoopman, 2016-12-27:
While yes they need to be similar you still want to be able to say which material a section should print in so it knows which extruder to use right?
It would be fine if in the settings section it only shows 1 material for temp control and all, but when editing the bed layout still need to be able to select else it will only ever print with the one extruder.

unlimitedbacon, 2016-12-27:
Ah, I understand now. You are referring to the material selection in the part editor, not the slice settings. I can confirm this issue is happening in 1.6.1, but not in source. This seems to have been fixed already.

closed 2017-02-01 by unlimitedbacon