MatterHackers/MatterControl #1761: [layer_count] variable

maukcc, 2016-12-28:
In the beginning of the gcode file, there is a line:
; Layer count: 203

Could you make this a variable to put in custom gcodes?
like in layer change gcode:
M117 printing layer [layer_num]/[layer_count]

tcm0116, 2017-01-13:
I was looking at this a bit because I would like the same function. It turns out that the existing [layer_num] variable is handled directly in MatterSlice and not in MatterControl. Since MatterSlice actually generates the layers before it creates the G-Code, then it should be possible to have it process a [layer_count] variable as well. The only issue that would need to be addressed is that raft layers are negative, so it would need to be decided if [layer_count] should be the sum of all of the layers including the raft, or if it would be better to have a [max_layer] variable that just contains the index of the top-most layer.

carlosen14, 2018-04-13:
Is there any update in this? I'm facing this too. I would like to implement a, kind of, progress bar or at least similar to @maukcc .

I think any of the two solutions you did provide will help me, because i mostly just want to get the percernt of advance. 

Thanks.

unlimitedbacon, 2018-04-16:
Progress reporting on the printer's LCD (either using `M117` or `M73`) has been added for MatterControl 2.0.
https://github.com/MatterHackers/MatterControl/pull/3075

There's no plan to add the custom G-Code variables, but if you still want them you are welcome to implement them yourself. It shouldn't be too hard.

unlimitedbacon, 2018-04-17:
Not yet. It is in alpha. If you want to try it you can either build it from source (the design_tools branch). Or download a Windows build from https://mattercontrol.appspot.com/downloads/matterhackers-experimental-windows/development.

carlosen14, 2018-04-17:
Thanks for the info.

closed 2020-06-07 by larsbrubaker