MatterHackers/MatterControl #1765: Can't export files to SD card on T10

unlimitedbacon, 2016-12-29:
Trying to export any file to the external SD card fails. Same with USB drives.

Error message:
```
Access to the path "/data/data/MatterControlAndroid.MatterControlAndroid.T10/files/.local/share/MatterControl/Library/at24dfqd.stl" or "/storage/extsd/Rocket_Shot.STL" is denied.
```

Other apps (like the file manager) are able to write to the drive without trouble.

jlewin, 2017-01-12:
Confirmed the problem repros on T10. Wrote a few simple tests that attempt to read and write from `/storage/extsd/existingfile`. The reads work just fine but code that writes to `/storage/extsd/filename` fails. Also confirmed I could write to `/storage/extsd` via ADB so I know the path is valid and writable at some level.

Looking online it appears writing to removable SD cards is prohibited in Android 4.4+ (https://metactrl.com/docs/sdcard-on-kitkat/). Further, there's some mechanism that requires significant user effort to enable access to external storage on Lolipop (TX7 Only) but it looks like a total nightmare to support and configure (https://metactrl.com/docs/sdcard-on-lollipop/).

Unless I'm missing something, it seems like writing to external storage is not currently possible. Would love another set of eyes to confirm

jlewin, 2017-01-23:
For now we should wrap the write call and warn on failure

larsbrubaker, 2017-01-23:
Warning is already happening.

**Looks like we can modify platform.xml to get premission**

Go to /system/etc/permissions/
edit platform.xml

add:
<permission name="android.permission.WRITE_EXTERNAL_STORAGE" >;
<group gid="sdcard_rw" />;
<group gid="media_rw" />;
</permission>;

jlewin, 2017-02-17:
As an alternative to modifying the firmware, we might look into use the Android file pickers... as previously discussed