MatterHackers/MatterControl #1780: Retraction errors leading to fast moves across parts

ke9g, 2017-01-04:
My understanding is that if a Z lift is specified in the filament retraction settings it will always lift when there is a non-extrusion move for distances greater than the prescribed 'minimum travel requiring retraction'.  But that is not always happening. I am getting retractions as expected about 75 percent of the time but then sometimes there are rapid moves across a part or multiple parts without a Z lift and the nozzle is hitting parts hard at high speed which sometimes leads to parts coming loose from the bed.  I looked through the gcode on one part and included a snippet below.  What I don't understand is why there are rapid G0 moves without extrusion that are not lifted from the layer height.  These rapid G0 moves don't seem to be adding any value to the part.  'wipe before retract' is not being used.

 I am using MC 1.6.1 on MacOS 10.11.6.

Retractions seems to be happening OK for the first few layers, but the erroneous non-printing rapid moves seem more prevalent in later layers.

Gcode attached for the first few layers of a collection of parts.

Here is an example, starting at line 18395. Comments below are mine. The ‘errors’ seem to be related to G0 commands where there are rapid moves at layer height over long distances.  This is what is knocking parts off the bed.  I would prefer a Z lift for any moves between parts.

G0 F12000 X12.296 Y-5.298 ;fast move with no extrusion???
G1 F3000 X18.509 Y-16.058 E3945.32282 ;normal line
G0 F12000 X18.51 Y-16 ;fast move with no extrusion ???
G1 F3000 X5.264 Y-16 E3945.76338 ;normal line
G1 F1500 E3941.76338 ;retract filament
G1 Z6.1 ;lift nozzle 4 mm
G0 F12000 X18.211 Y-16.456 ; fast move to new XY but this is OK because nozzle is up
G1 Z2.1 ;drop nozzle 4 mm
G1 F1500 E3945.86338 ;restore filament position
G1 F3000 X18.509 Y-15.94 E3945.8832 ;normal extrusion
G0 F12000 X14.093 Y4.598 ; fast move without extrusion??? Y distance is 20 mm
G0 X14.893 Y5.14 ;another move without extrusion???
; TYPE:WALL-INNER
[parts1.gcode_excerpt.txt](https://github.com/MatterHackers/MatterControl/files/683811/parts1.gcode_excerpt.txt)

stephenci, 2017-01-15:
Same issue here. Flag as important bug please.
I thought I have retraction issue causing unusually lots of stringing turns out OP described the exact issue I'm having.

larsbrubaker, 2017-01-18:
Thank you. Our guys will look into it and document what we need to do.

unlimitedbacon, 2017-02-01:
Could either of you post your printer profile and the STL you are experiencing this issue with? Some work has been done on the retraction code recently and I don't think this is still an issue, however I want to test with your settings to make sure.

stephenci, 2017-02-01:
Mine ia the popular 3D benchy...

duisenberg, 2017-02-05:
As for 1.6.2.8214 I can confirm it still is an issue. Having set Minimum Travel Requirement to 0 and Minimum Extrusion to 0.1mm Retraction/Z-Hop still sometimes does not fire leading to scratch marks. 
Seems to be similar to Cura, there it happend all the time, preferably ruining the top layers.

duisenberg, 2017-02-09:
To fence the issue in a bit more - the missing "Z-Hop/Retraction" seems to occur only in printing infill parts of the bottom/top layers. Infill in general is printed in parts, starting randomly somewhere and then printing to perimeter or an edge. Then comes a travel move to the start of next infill part but this one is done without Retraction/Z-Hop, leading to scratch marks if the nozzle runs over the already printed infill part. Depending on the structure this may kick the model from platefor bottom layer, for top layer it is just some ugly scratch marks.

Saphir17, 2017-02-19:
I agree to the top post by Ke9g.

I get weird moves without retract and Z-lift, where it is obvious that it should have been a retract.

MC 1.6.2.8214, Orion Delta.

larsbrubaker, 2017-02-22:
I would love to know if you are seeing the same thing on the latest 1.7 alpha. If you switch to the alpha channel

larsbrubaker, 2017-02-22:
Thanks for the feedback.
Can you please post a picture, your profile, and the stl that is causing the issue and we will look into it.

If you can use the newest 1.7 alpha that would be great as well. It is currently up for Windows.

ke9g, 2017-02-23:
Thanks, but I slice on a MacBook Pro.  Please let me know when that alpha is available and I’ll give it a try.

/Tim

Saphir17, 2017-02-23:
Attached are STL, Layer view and a picture of a printed part.
This part does some moves from the top from layer 1 to layer 43.
This move do not retract/z-lift.

[P38_fus_center_01.zip](https://github.com/MatterHackers/MatterControl/files/797460/P38_fus_center_01.zip)

[image]

Saphir17, 2017-02-23:
Ups forgot the picture.
As you can see a lot of stringing from other areas too where retracts do not happen either.
The mentioned move is the move from top to center (almost center)

[image]

Saphir17, 2017-02-23:
A highlighted snip of the move that among others do not retract.
[image]

unlimitedbacon, 2017-02-23:
@Saphir17, please export and post your slice settings file as well.
http://wiki.mattercontrol.com/Send_Debugging_Information#Slice_Settings

duisenberg, 2017-02-23:
Using 8347, at first glance travel/Z-hop shows improvement. 

At start for skirt it hops/wipes at the end of each round. fun move ;) 

With infill bottom/top it
- bottom first layer is now done in "one move" without split 
- bottom / top intermediate layers are split using Z-hop on travel most times 
- top layer still suffers from travel without retraction/hop  

All in all: it compared to 8214 looks better but not yet done.

Attached ZIP might point to various issues. Contains STL and printer settings. 

[Travel issue 1780.zip](https://github.com/MatterHackers/MatterControl/files/797719/Travel.issue.1780.zip)

The partial a bit messy flow in the images seems to be related to the print material which seems to become oozy and to suffer from diameter issues on the second half of the spool. 

[image]

Above the scratch mark is quite obvious

[image]

The stl has been scaled down for a quick test print. To work it must be Y>130mm.

Saphir17, 2017-02-27:
Hey UnlimitedBacon...

The Slicer setting for my Orion Delta.
Have been fiddling with temp, speed, retraction but never got it better the the above picture.
Most of the stinging relates to moves that do not retract. When it retract, the move do not string.

[MatterControl slicersetting - Orion Delta.zip](https://github.com/MatterHackers/MatterControl/files/804676/MatterControl.slicersetting.-.Orion.Delta.zip)

Saphir17, 2017-02-27:
Installed MC Alfa version 1.7 and it actually printed a lot better, but there is still some moves that do not retract.

After the 1.7 print I reduced the extrusion multiplier from 1.15 to 1, with a at worse result ??

The three print aligned in two light settings : 
From left to right : 
v1.6
v1.7 alfa
v1.7 alfa, extrusion mult. =1

[image]
[image]

duisenberg, 2017-02-27:
Same here, with 8375 the scratch marks seem to be gone, still a (very) few spider webs left. 

In my case I put the spider issue mainly to the material that seems to have diameter issues in the second half of the spool. Had to increase flow to 1.2. Also my  MK8 printhead seems to have a gap between nozzle and threaded rod in the heater block, leaving dim dead volume that buffers retraction/extrusion. Only minimal, but may lead to an a tiny bit late extrusion after retract/travel.

duisenberg, 2017-02-27:
Hey Saphire17 - did u check the stl in netfabb? There seem to be some issues with the model itself, maybe these confuse matterslice too.

Saphir17, 2017-02-28:
Hi Duisenberg...

No, haven't tried checking the STL. Will do...

Interesting if I have a diameter issue with the material. I will go back to 1.15 extrusion or more to see if it cleans up even more.

The left side spiderweb in all three prints comes purely from a move with out a retract. I can see move without the "woop-woop" sound the retract makes.

A while ago I wonder if gravity had anything to do with the oozing too, so I turned the printer up-side down.
It printed well... but kept oozing... upwards.

Btw. the print I posted is the nose-section of a model-plane from 3DLABPrint, The P-38. Check it out.