MatterHackers/MatterControl #1790: No retraction at the end of a layer

mc71, 2017-01-05:
MC for Me3D **1.6.0.7973**
There is a small bug in the identification of islands or perimeters.  (The part is one of the 3Dprintlab aircraft components, all single surfaces).

There is no retraction between start of a layer and the end of a previous layer when there really should be based on the settings. This is causing a break in the skin of the single walled object.

[image]

Retraction working with "Avoid crossing Perimeters" on
[image]

[image]

larsbrubaker, 2017-01-06:
Great pictures and bug. Can you attache the profile and stl in a zip?

Thank you.

fastslash25, 2017-01-07:
I too am consistantly having the same issue.
[HyperCube.docx](https://github.com/MatterHackers/MatterControl/files/691246/HyperCube.docx)
http://www.thingiverse.com/download:1481537
I could not attach .zip files for some reason. So the profile is in a doc file and theres the download to the .stl.
The stl is scaled to .7667 scale.
[image]

mc71, 2017-01-09:
Hi @larsbrubaker follow this link for an example.
https://drive.google.com/drive/folders/0B9RYfOH4aNfKMkkyVGJKSDEyN2c?usp=sharing

larsbrubaker, 2017-01-31:
Thanks for the report.
This is fixed in master (for both reports)

The first profile had an error that the external perimeter was 0. The new version will report this error.
We'll also write a test to check if our profiles have this set incorrectly.
We're working to get these changes out soon.

[image]

[image]

closed 2017-02-06 by larsbrubaker