MatterHackers/MatterControl #1799: Importing printer does not affect OEM layer

indrora, 2017-01-06:
I've built my own printer. When trying to make sure the "defaults" are hammered down, I've exported my printer out and edited it. 

As a zen-like example, I've edited `.OemLayer.print_center` to a specific position. I import the profile back into MatterControl and when I check "printer -> print area -> print center", I get that point with the reset marker next to it.  Pressing "Restore Default" restores to 100,100 -- which in my case is very wrong.

this implies that importing a profile doesn't change the defaults. I need to be able to edit the defaults without mucking about in user files.

larsbrubaker, 2017-01-13:
Thanks for the report. We have an internal issue we are tracking with this. We will make the option to bake user settings into the profile and still preserve the ability to get back to the original Make Model when required (not what you are doing, but important for many users).

Adding our internal bug for our devs to see with this one.
MatterHackers/MCCentral#977
Add way for users to bake UserLayer into OemLayer

larsbrubaker, 2018-09-24:
Closing - fixed.

Please file a new issue if error persists.

We added the ability to 'set as default' on right click a user override.

closed 2018-09-24 by larsbrubaker