MatterHackers/MatterControl #1814: Updater is broken on Mac

whosawhatsis, 2017-01-11:
I just tried to update (sorry, I can't tell you exactly which version I had, because it's gone now) from within MatterControl. MatterControl told me there was an update available, and the updater appeared to run, but it instead deleted mattercontrol. It's not even in the trash, I had to re-download from the website.

whosawhatsis, 2017-01-17:
It just did it again, so not fixed in whatever version I had. I'm on the Alpha channel, btw. Kevin indicated that that might matter.

unlimitedbacon, 2017-01-19:
Was unable to reproduce this when upgrading from 1.6.0.7857 to 1.6.2.8242 on macOS 10.10.5. Nevertheless, I've heard of this happening to other people in the past. I think the best solution would be to stop using the installer wizard and move to .dmg packages like most other Mac applications.
https://github.com/MatterHackers/MatterControl/issues/1256

unlimitedbacon, 2017-01-20:
The installer works by looking for an existing copy of MatterControl on your system and overwrites it. If there is another copy of MatterControl anywhere on your system, it could be upgrading that one by mistake and deleting the one in your Applications folder. This includes any copies that might be in the trash. You can use Spotlight to search for other MatterControls and remove them.

ArtDezign, 2017-03-20:
This reproduces on my Mac OS X(10.12.3) running MatterControl 1.7.0

larsbrubaker, 2018-11-15:
This should be fixed as of 2.0. We now use DMGs.

closed 2018-11-15 by larsbrubaker