MatterHackers/MatterControl #1815: What happened to gcode flavor?

whosawhatsis, 2017-01-11:
In 1.6.1 (build 1.6.1.7992), the option for gcode flavor appears to be gone. What happened, and how can I access this setting?

larsbrubaker, 2017-01-12:
What is it that you were needing in there? I had no tests for if these were working and no printers on site that needed them. I was printing with the bukito for a long time without changing the flavor.

Can you get it to work with the standard reprap settings?

whosawhatsis, 2017-01-13:
The problem is that after an update, we got strange results, and as part of the process to diagnose the problem, I tried to check which gcode flavor was being used, but the setting was nowhere to be found.

The setting needs to exist somewhere. I could see moving it to the machine setup (next to serial port selection), but it wasn't there, either.

The best results with a Bukito are achieved in volumetric mode, with E units of cubic mm and firmware retraction, in the mode that mattercontrol mistakenly calls Ultigcode (Both of those things are true of Ultigcode, but that's not all there is to it). We previously had MatterControl configured to operate that way. We use a start code that uses an M200 to put extrusion in volumetric mode, but MatterControl seems to have downgraded it to normal reprap gcode, with mm E units and no firmware retraction. This does not work at all with our start code, and gives inferior results if we modify the start code to make it work.

Tannoo, 2017-01-13:
My 2¢:

I tried MC for the first time and setup using MatterSlicer and Prusa i3 settings. Well, the printer printed the part just fine, but all kinds of weird info was going on.

Octopi couldn't interpret the layers correctly. While it was printing, the layer numbers were displayed randomly and erroneously. Meaning, it looked like it was rolling die for the layer number after every move. Erroneously....meaning, total layers were 447, but the random layer numbers were from 0 to ~3500.

Temps were reporting correctly, but watching the Gcode in Octopi was useless.

When printing from MC directly, the marlin firmware couldn't decipher certain things correctly. The LCD display wouldn't display that the bed was heating. But, when the hotend was heating, it did say "Heating...". And the display never changed it's status from that for the entire print.

I've run into weird things like this when the 'flavor' has been set incorrectly in Repitier, but this is my first time using MC.

unlimitedbacon, 2017-02-03:
@Tannoo, I'm not sure how Octoprint keeps track of the layers, but if it is looking at the Z axis moves then software print leveling would make it go crazy. Z lift probably isn't helping either. Is there some kind of tag we could add to the gcode to help it identify the layers correctly?

MatterControl intercepts `M109` and `M190` commands and converts them to `M104`/`M140`. These commands are probably what trigger the heating messages on your LCD. You could look into using `M117` to display whatever messages you want, and insert it as appropriate into your [Start G-Code](http://wiki.mattercontrol.com/SETTINGS/Printer/Custom_G-Code/Start_G-Code).

Tannoo, 2017-02-03:
From what I see, any command to move Z triggers a layer change. In Repetier and in Octoprint.

Why does MatterSlice have soo many Z moves? Here is an excerpt from the MatterSliced code.
This section is just for the skirt:

```
; TYPE:SKIRT
G0 F1800
G1 X159.6 Y14.27 Z0.128 E0.815
G1 Y170.73 Z0.216 E7.32
G1 X145.73 Y184.6 Z0.23 E8.136
G1 X53.27 E11.98
G1 X39.4 Y170.73 Z0.228 E12.795
G1 Y14.27 Z0.273 E19.3
G1 X53.27 Y0.4 Z0.27 E20.116
G1 X145.74 Z0.14 E23.96
G0 X145.57 Y0.8 Z0.141 F7800
G1 X159.2 Y14.43 Z0.129 E24.762 F1800
G1 Y170.57 Z0.216 E31.253
G1 X145.57 Y184.2 Z0.229 E32.054
G1 X53.43 E35.885
G1 X39.8 Y170.57 Z0.227 E36.687
G1 Y14.43 Z0.273 E43.178
G1 X53.43 Y0.8 Z0.27 E43.979
G1 X145.57 Z0.141 E47.81
G1 E46.81
G0 X56.08 Y177.8 Z0.225 F7800
```
This is the CuraEngine slice of the same file (skirt only):
```
;TYPE:SKIRT
G1 F1800 X55.175 Y3.800 E0.55856
G1 X144.825 Y3.800 E3.54033
G1 X156.700 Y15.675 E4.09890
G1 X156.700 Y169.325 E9.20932
G1 X144.832 Y181.193 E9.76755
G1 X142.925 Y181.200 E9.83098
G1 X55.175 Y181.200 E12.74956
G1 X43.300 Y169.325 E13.30812
G1 X43.300 Y15.675 E18.41854
G1 F2400 E15.41854
G0 F9000 X47.100 Y17.248
```

Tannoo, 2017-02-03:
I did print the MatterSliced version of the file in question. MatterSlice generated the proper supports, especially where Cura just seemed to overlook. Slic3r, for some reason sliced it no better than Cura.

Slic3r also generated code similar to Cura.

unlimitedbacon, 2017-02-03:
The Z moves are how the software print leveling works. In order to account for the different heights of different positions on the bed MatterControl is constantly lifting the nozzle up or down as it moves. Turning off software leveling will eliminate the Z moves.

Repetier and Octoprint are operating under the erroneous assumption that a Z move indicates a layer change. Unless there is some other way to indicate the current layer number to these programs, there is not much we can do.