MatterHackers/MatterControl #1821: Unable to use Slic3r or CuraEngine

tcm0116, 2017-01-13:
Fresh install of 1.6.1 on OS X 10.11.5. Did some setup and was able to slice with MatterSlice without issue.

However, when switching to Slic3r, I get "Slicing error. Please review your slice settings." I copied the Slic3r package out of the MatterControl package so I could run it stand-alone and attempted to load the ini file that was generated, which resulted in an error: "I was unable to load the selected config file: illegal division by zero at /Applications/Slic3r.app/Contents/Resources/lib/std/Slic3r/GUI/Plater/2D.pm line 256.". Attached is the configuration ini file that is generated by MatterControl that is causing the issues.

Next attempt was to use CuraEngine. When I press Generate, I just get the "Press 'generate' to view layers" message, but no other errors or status. I couldn't find any troubleshooting steps for CuraEngine, so I'm not sure where to go from here.

[config_3511736756072661045.ini.txt](https://github.com/MatterHackers/MatterControl/files/703246/config_3511736756072661045.ini.txt)

unlimitedbacon, 2017-01-25:
Confirmed. Slic3r is not working at all on Mac build 1.6.2.8242. When running the slic3r executable from within Mattercontrol.app it fails with the following error.
```
Failed to initialize virtual file system at <Packaged Modules>/Application::Loader line 1183.
BEGIN failed--compilation aborted at /Applications/MatterControl.app/Contents/MonoBundle/Slic3r.app/Contents/Resources/lib/std/File/Glob.pm line 7.
Compilation failed in require at <Packaged Modules>/Application::Loader line 1433.
```
However, after I take Slic3r.app out of MatterControl.app it works fine.

jlewin, 2017-02-16:
Both should be working in the latest experimental builds. 

@unlimitedbacon - Can you take a look and confirm this problem no longer repros?

unlimitedbacon, 2017-02-18:
Negative. Neither Slic3r nor Cura are working on the Mac in the lab. Build # 1.6.1.8295.

jlewin, 2017-02-23:
Looked into this yesterday and again today. I typed up a response but I'm not seeing it here so I'll briefly recap.

I too received the virtual file system error:
```Failed to initialize virtual file system at <Packaged Modules>/Application::Loader line 1183...```

Interestingly when run under sudo as described in https://github.com/alexrj/Slic3r/issues/2020, Slic3r launched without issue. It seems likely this is a permissioning issue and the Slic3r bug remains open.

The MatterControl package seems to work everywhere but on the lab machine. I blew away the local package and after reinstall I'm no longer seeing the error

closed 2017-03-04 by larsbrubaker