MatterHackers/MatterControl #1824: Locks up on print everytime after v1.5.0 (still holds temp)

midnightcharger2, 2017-01-16:
Love mattercontrol, Gui layout just works great.

Using a Macbook air, but but issue shown up on windows also
Mks-sbase (v1.1 and also v1.3) with 1/32 stepping (current for all motors is 0.85a for testing) on kossel mini delta

Any version after v1.5.0 the steppers lock up and hotend and heatbed still keeping temp and fan still spins.  This usually happens on the first layer on the skirt.  Looking at terminal, it is still sending Gcode, and Smoothieware status LEDs are blinking as normal.  Computer does not disconnect from the board, but cancel print or any other action is ignored.  Needs a hard reset on the board each time.

What I have noticed is if I keep the terminal window open while printing, it seems to get further in the printing process (still locks up).  If I attempt to close terminal while printing, lockup happens almost immediately afterwards, or within that layer.  

Thought it was an EMI issue or USB and even swapped out boards, but each board and Usb cable produced the same result.  Tried with another slicer software and no printing issues.  

Just reversed through the list of each version of mattercontrol and v1..5.0 has not locked up once.  Testing now and air printed the same .stl 5+ times without issue....

unlimitedbacon, 2017-01-18:
Could you post a [terminal log](http://wiki.mattercontrol.com/Send_Debugging_Information#Print_Log_.2F_Terminal_Log) from when this happens?

Could you also try updating to the latest alpha version?

midnightcharger2, 2017-01-18:
Sure thing on the terminal log, most likely this weekend due to work commitments.  Quick update though.  From what I noticed with the recent Alpha release (compared to the recent "stable release", it prints, no lockups!!  Thanks for the advice ULB, but introduced a some serious stuttering.  Most noticeable when homing to Z0 to print.  When printing, the same stl, there was noticeable stutter points around the outside shell of the print at the exact same locations on each level.  Not 100% sure, but if it was stepper cross-talk, it would be random and happen on v1.5.0 also.  Let me establish a strong baseline again with v1.5.0 and hardware first where I know I can get consistent prints time after time and then can guinea pig the machine again to send the terminal log for both the "stable and Alpha releases"  I could also video it as to help visualise what is going on.

larsbrubaker, 2017-02-21:
Thanks for the follow up. I'm closing this assuming it is fixed. Please let us know if not. Also, there is a new alpha!

closed 2017-02-21 by larsbrubaker