MatterHackers/MatterControl #1831: Before Tool Change g-code, does not insert the code added at material level

MERLINCODE, 2017-01-18:
Version 1.6.2.8176
2 extruders
share temperature is un ticked.
Trying to add before and after tool change g-codes, at material level..
After tool change g-code adds code perfectly.
Before tool change g-code inserts nothing.
Need this feature to set tool operating and standby temperature:
e.g. M104 S170 T0; M109 S205 T1;
Please can this be fixed, or give me a workaround?
Thanks

unlimitedbacon, 2017-02-01:
Seems to be working on the latest stable release. Example gcode file included. 1.6.2.8176 is an older build. Can you update and see if it is still a problem for you? If so, please [export and post your slice settings](http://wiki.mattercontrol.com/Send_Debugging_Information#Slice_Settings).

[2073006765381805892_2_-7515689457622545622.gcode.txt](https://github.com/MatterHackers/MatterControl/files/746304/2073006765381805892_2_-7515689457622545622.gcode.txt)

larsbrubaker, 2017-02-21:
Closing - assumed fixed.

closed 2017-02-21 by larsbrubaker