MatterHackers/MatterControl #1840: G2/G3 arc creation from .amf

nokiann, 2017-01-20:
In openscad I created a round shape, exported it as .amf file, imported it in mattercontrol and exported it as gcode file.
```
// rotate_extrude() always rotates the 2D shape 360 degrees
// around the Z axis. Note that the 2D shape must be either
// completely on the positive or negative side of the X axis.
color("red")
    rotate_extrude()
        translate([10, 0])
            square(5);
```

Looking at the gcode file I can't find the G2 or G3 arc commands
http://reprap.org/wiki/G-code#G2_.26_G3:_Controlled_Arc_Move

What's the advantage of using .amf if the generated gcode is the same?

larsbrubaker, 2017-01-20:
Amf has many benefits
1. It stores redundant vertices as single items so it is smaller
1. It can use compression, so it can be even smaller
1. It can have multiple objects in the scene
1. It supports some advanced features, like curves and subdivide surfaces

However, open SCAD is only supporting same vertices.

You can actually open the .amf file that OpenSCAD saves (it is just a text file) and see that it is just the polygon data.

------------------

Now, with all that said, it is true that MatterControl does not currently support curves. There is lots of ambiguity as how to support them, and like OpenSCAD few tools output them at all, so there is also little practical value.

nokiann, 2017-01-20:
Alright so I assume the main task is creating a robust arc detection algorithm.
The advantages don't sound useful for the usual 3d printing applications.
I think the main reason for .amf is to get more accurate hole dimensions.

closed 2017-01-23 by larsbrubaker

adnion, 2017-03-05:
FreeCAD-git supports .amf export.
OpenSCAD does.
SolidWorks does.
Rhino3D does.

I thought .amf -> G2/G3 arc commands will help people to get tighter tolerances when printing circular things...