MatterHackers/MatterControl #1853: STL not processed correctly

tcm0116, 2017-01-26:
Using MatterControl 1.6.2.8214, I'm having issues with the STL file below.

[CableTie.stl.txt](https://github.com/MatterHackers/MatterControl/files/732836/CableTie.stl.txt)

When slicing with MatterSlice or CuraEngine, portions of the part are being excluded. When using Slic3r, the infill on the top layer is not filled in. When slicing the same STL file with Cura or Slic3r, there are no issues.

[image]
[image]
[image]
[image]

unlimitedbacon, 2017-01-26:
Inspecting this mesh in Blender shows that it has a lot of intersecting and non-manifold geometry. I'm not surprised that it is having trouble slicing. I agree that MatterSlice could do a better job of handling bad meshes like this, but in the mean time you can try running it through a service like [NetFabb](https://service.netfabb.com/login.php) or [MakePrintable](https://makeprintable.com/) to clean it up.

tcm0116, 2017-01-26:
I was thinking that may be the case. However, why does it work in the Cura
and Slic3r applications and not in MatterControl when using those slicers?

unlimitedbacon, 2017-01-26:
I believe Slic3r does it's STL repairs when the file is first imported, not when the slicing is done. Using the Slic3r engine directly (as MatterControl does) would bypass this step. However, it looks like we could add this abitity by running `slic3r --repair` before the slicing operation. http://manual.slic3r.org/advanced/command-line

 Cura probably works in a similar way.

tcm0116, 2017-01-26:
Perhaps it might be worth looking into adding a repair feature to
MatterControl such that the STL passed to whatever slicer would have a
better chance of succeeding.

larsbrubaker, 2017-01-31:
Thanks for the feedback. Making better repair is a separate issue that is in our feature list. Closing this for the time being.

Adding another issue for slic3r repair.
#1865

closed 2017-01-31 by larsbrubaker