MatterHackers/MatterControl #1856: Copying a mesh is very slow

DanielJoyce, 2017-01-28:
Copying a mesh is VERY SLOW in mattercontrol. A 5mb mesh takes 20-30 seconds to copy. It frankly boggles my mind how can it be so slow.

This model is only about 5Mb a piece

http://www.thingiverse.com/thing:1011980

"Generating GL data" seems to take a long time. Are you guys not using basic instancing with your meshes?

uname -a

Linux crusoe 4.8.0-34-generic #36-Ubuntu SMP Wed Dec 21 17:24:18 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

cat /etc/lsb-release

DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.10
DISTRIB_CODENAME=yakkety
DISTRIB_DESCRIPTION="Ubuntu 16.10"

larsbrubaker, 2017-01-31:
Thank you for the report. This is fixed in a new branch we are working on that improves our editing tools (the branch is called design_tools if you want to check it out).

Closing this at this time and hopefully we'll have the new features out soon.

closed 2017-01-31 by larsbrubaker

DanielJoyce, 2017-02-02:
Super duper faster in that branch! Works great. Scale and move controls are a bit broken. Will see if I can track down to fix. As project file references seem broken, had to re add a whole mess of them to get them working in monodevelop.

DanielJoyce, 2017-02-03:
Switched to vs-code. Inter project and other references still seem broken though compiles fine.